jameszah/ESP32-CAM-Video-Recorder #2: How to get deep sleep working?

M0ebiu5, 2020-01-04:
I'm trying to wake up the esp32 with a pir, but it keeps waking up, even with no connection on the pin.

The same pir is working with another sketch, so i assume it has something to do with the recorder code? 
I'm using this in the loop function:
```
if (recording == 0) {
    Serial.println("Entering deep sleep mode");
    Serial.flush(); 
    esp_sleep_enable_ext0_wakeup(PIR_PIN, 1);   //wake up when pin 12 goes HIGH
    esp_deep_sleep_start();  
  }
```

jameszah, 2020-01-04:
GPIO12 is HS2_DATA2 used to send data to the SD card.

I'm not sure if there is a free gpio wired to the two 8 pin headers.
GPIO 33 is used for the red led on the back, so maybe you could use that with a bit of soldering.
Or re-use the rx and tx???  Maybe someone on the web has a solution

Here is the schematic.   
https://github.com/SeeedDocument/forum_doc/blob/master/reg/ESP32_CAM_V1.6.pdf

M0ebiu5, 2020-01-04:
Thanks for your hints! I just realized that there are 2 different SD card interfaces and libraries!
I was using the SD library before - i guess it has lower performance than SD_MMC...

Will try the suggested other pins (i need 2, cause the most usefull iteration should be able to measure the battery voltage).

closed 2020-01-04 by M0ebiu5

jameszah, 2020-02-25:
new pins are available!