jameszah/ESP32-CAM-Video-Recorder #3: recording video without internet

amirjak, 2020-02-16:
It is possible to use ESP-cam without internet to start recording video and stop.
 for example: add  key (LOW or HIGH)

jameszah, 2020-02-16:
Hi,

I think if you removed the major_fail() call in the init_wifi() and init_time(), then it would continue running and record with initial conditions, and just date everything with 1970.  And continue until it filled the sd card.

The major_fail() flashes S-O-S on the little red led, and then reboots the ESP.

A more general solution with a button or PIR etc, needs another GPIO pin, and there are none available.  There are 6 gpio for the SD card, 1 for programming the device, and 1 for controlling the PSRAM - I think that is true, although I'm not sure why that is wired to a header???

If you can give up the comfort of debugging messages on the console -- the Serial.begin(115200) and Serial.println("hello world"); then those 2 gpios could be used to attach a button or PIR, etc.

I've been meaning to give that a try.

amirjak, 2020-02-17:
Thanks for your quick response

amirjak, 2020-02-18:
I found this one.

https://github.com/s60sc/ESP32-CAM_MJPEG2SD

Ground pin 12 with breadboard wire, female start recording video and Remove stop.

jameszah, 2020-02-18:
Hmmm ... very nice!

I was just yesterday trying to understand the 4-wire vs 1-wire io system.  The 1-wire does not make things much or any slower -- I do so many write to put the jpeg into avi format, that the greater speed of the 4-wire write does not give me much benefit -- and it uses those 3 other pins that could be used for other things.

I think that program just stuffs all the jpegs together in a file, and then pulls them apart to present them as jpegs -- rather than preparing an avi file that your pc can play without pre-processing.

It has some very nice code -- efficient writing in big 1-MB blocks, etc.

I'll see if I can make a PIR or microwave-sensor version using the 1-wire system in a day or two.

I thought I'd just record everything, and if there is no activity in the last 30 seconds, then close the file and delete it.  If there is activity, just keep recording until activity stops for 30 seconds.

s60sc, 2020-02-21:
If you want more pins:
IO0 can be used if what you attach does not pull it low at startup, eg a button
The blink led IO33 can be re-purposed by removing the led and some soldering - I use it for an analog LDR to control the lamp led.
UOR (IO3) can be used for input depending on what circuit you attach. I use it for a DS18B20 temperature sensor - it does not interfere with either serial port output or flashing
UOT (IO1) could be used for an output but I haven't tried.

amirjak, 2020-02-23:
Thanks for the help.
 I need to remove all internet (wifi) code. 
No browser. 
Default resolution VGA. 
No date and time. 
On code record by number if avi1 exists Make avi2 (...). Just a button to record and stop. 
Is that possible.

jameszah, 2020-02-23:
Yep - no problem.

I'll do a version to control with a pin, and then get the sd chip to see the results.

jameszah, 2020-02-25:
new version - give that a try, see if it works

amirjak, 2020-02-28:
Thanks, it's working.

closed 2020-03-05 by amirjak