jameszah/ESP32-CAM-Video-Recorder #5: why do_fb() should run 5 times？

asjdf, 2020-03-05:
is it necessary？and would you mind telling me how to change camera mode (turn rgb to grayscale and change framesize)?i had read your code but i still have some question.

yours,
Atom

jameszah, 2020-03-05:
I think the newer version just has just 3 discarded frames with a 1 milli-second in between.

I think you need at least one, or the first frame will be a little scrambled as the camera is setting its new characteristics.  And I had some problems with the camera needing a second of two to read the light and set the exposure correctly.

So you could try 1, and check the first frames of your recording to see if they look okay.

jameszah, 2020-03-05:
Some parameters are set for the hardware and the camera drivers before the esp_camera_init(&config); is called, which sets up the hardware and the camera drivers ... memory allocation in particular.

After that, you can change some parameters with the code below.  There are loads of other parameters (white balance, brightness, contrast, etc) that can be set in a similar way.  You can find all those in the demo program.  These are the only ones I was interested in.

sensor_t * ss = esp_camera_sensor_get();
  ss->set_quality(ss, quality);
  ss->set_framesize(ss, (framesize_t)framesize);
  if (gray == 1) {
    ss->set_special_effect(ss, 2);  // 0 regular, 2 grayscale
  } else {
    ss->set_special_effect(ss, 0);  // 0 regular, 2 grayscale
  }

asjdf, 2020-03-07:
Thank you very much.

By the way,will mobile detection be added in the future?（i am looking forward to this feature) : )

jameszah, 2020-03-07:
mobile detection?   Do you mean motion detection?
Motion detection is there in folder /v60   
It arrived about a week ago.

asjdf, 2020-03-08:
yeah motion detection.i just copy the words from the translation program and didn't check it.ahh
thank you.

asjdf, 2020-03-08:
emmm.....I mean the kind of motion detection that is achieved with image recognition, not the kind of infrared thermal sensor.Is it possible?

jameszah, 2020-03-08:
I've seen a couple people doing that on github.  It usually involves making the frame very small and comparing two frames, then deinit() and init() with a large frame to take the picture.  I have not had much luck with deint() and init() -- it seems to fail frequently.  

I'll have a look at it again, to see if I can do it without the deinit() and init().

closed 2020-03-11 by asjdf