jameszah/ESP32-CAM-Video-Recorder #6: Camera-based motion detection transformation successfully

asjdf, 2020-03-22:
I transformed and tested your project and successfully implemented motion detection.
Thank you in advance for this project.

But I currently face a small problem, which is often the "Queue Full" problem. My recording mode is 10fpsSVGA, while detecting movement every 100ms. Is it possible that I have touched the bottleneck of ESP32? Is it possible further optimization.

This is my code:[https://github.com/asjdf/ESP32CAMmotion_detection/blob/master/test2/test2.ino](https://github.com/asjdf/ESP32CAMmotion_detection/blob/master/test2/test2.ino)

jameszah, 2020-03-22:
Very interesting -- I'll have a look.

But 10 fps SVGA is very fast, even if there is no software motion detector.  
On my LEXAR 300x 32GB microSDHC UHS-I, I was only able to get 5 fps SVGA to work reliably, without getting into frequent "queue full" problems.

thomas-pegot, 2020-05-17:
You did both great job. 
I implemented myself a motion detection algorithm using the example from arduino CameraWebserver.
Here a little video : https://streamable.com/r1vt77
It seems slow because I manually take a still from the web interface and it computes (300ms).
It's a robust algorithm (optical flow) I might try to implement with this repo. 
Since it's optical flow I can remove any artifacts and unwanted motion such as leaf motion, birds, ..

thomas-pegot, 2020-05-20:
I will try to implement my library that I used in this repo : https://github.com/thomas-pegot/camera_web_server

DJprasenjit, 2022-10-04:
Is the send photo telegram works? In my case its not working