jameszah/ESP32-CAM-Video-Recorder #7: Is it possible to use web server instead of FTP?

kahoch, 2020-04-03:
For example, yi hack: https://github.com/fritz-smh/yi-hack
Everything can be done on browser, much more convenient.

jameszah, 2020-04-04:
I looked into that (http download) but couldn't find anything simple. 

You can click on the ftp://192.168.1.2 link and get a list of files and the directories, but when you click to download, it also says "network error".  I pursued that, and found that Chrome browser does not use the ftp RETR command, but the strange LIST -l command to download files.  I hacked that into the ftp code, but then I saw a thing that Chrome was abandoning ftp for security reasons.  And other browsers seem to use other sequences of ftp commands for downloading.

So I gave up.  A quick download of Filezilla ftp, and you can see a list, and download and delete all the little (or big) videos.  It seems little a practical solution for me.  And you don't have to right-click and save each file.

This guy seems to have some the code to do it, if anyone wants to pursue that.  It seems simpler than I remember when I looked at it the first time.

https://www.youtube.com/watch?v=u6vmUIPU1i4

kahoch, 2020-04-05:
Thank you for you work! Let me have a try on Filezilla.

bhyatyab, 2020-05-15:
You can access ftp by pasting below into Windows Explorer. Change esp_ip of ip of esp cam

ftp://esp:esp@esp_ip

closed 2021-01-21 by jameszah