jameszah/ESP32-CAM-Video-Recorder #10: sd problem

superjasiek, 2020-05-08:
Hi your project looks grate and it totaly fit my needs - i'm planning to build videotrap for birds feeder. anyway unfotunately after compiling v60 i'm getting sd error "can't" mount vfat. I use 8gb hc sd - it works with other esp32 cam projects. I would be gratefull for any advice.

jameszah, 2020-05-09:
Make sure pins 12,13, and 4 are not held down while your setup() and sd mount code is running.

I think pin 13 is the crucial one, but pin 12 and 4 might cause problems.

https://github.com/luc-github/ESP3D/issues/386#issuecomment-614358748

superjasiek, 2020-05-10:
Thank you for advice - a changed sd card (philips 32gb UHS-I and now it works fine.  I m testing the setup now. I've notice one thing - after some time the http interface becomes unavailable but recerding triggerd by PIR still works. Next step is going to power everything up with solar cell.

jameszah, 2020-05-11:
Interesting.
I have some cheap sd cards from the dollarstore that seem to need extra current to mount, and when they do mount, gpio 4 stays high after the mount, so the big led is on all the time. ??  

Regarding the wifi dropping out:  I have had trouble with wifi - make sure your Arduino esp32 library is up to date. I think 1.04 is current - previous ones had problems.

Using library ESP32 at version 1.0 in folder: C:\Users\James\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ESP32 

I'm fiddling with some solar cells and voltage regulators as well - not sure what battery to use.

superjasiek, 2020-05-11:
No, the big Led is fortunately off. Did you consider importing deep sleep mode into the code? I think that would be nice idea, considering energy saving. I'm trying to figure it out according to this article https://randomnerdtutorials.com/esp32-cam-pir-motion-detector-photo-capture/.

jameszah, 2020-05-12:
I think there are 2 applications here -- if you are filming 10 fps and keeping the internet connected for the http and ftp, then deepsleep is not an option.

But if it is 1 frame per minute timelapse, then you could deepsleep, and just save a jpg file, as I don't think an open avi file would survive the deepsleep.  (I think the internet has some examples of this.)

I suppose there is a third option of deepsleeping until you get a PIR event, then waking up, recording at 10 fps, and then sleeping again.  And you could have the internet turned on or off during that PIR event.  Turning on the internet is time-consuming, and power-consuming if your goal is to get a very fast recording of the motion.

I'm just rewriting this to separate the pieces so you could remove ftp or http or pir, .... and just take the pieces for your particular application.

superjasiek, 2020-05-22:
The idea of rewriting code to pieces sound grate - actually i do not need http interface . But i still struggle with energy consumption while recording video and looking for optimalization. The idea of go to sleep after record at 10 fps and wake-up on pir event sounds reasonable.

jameszah, 2020-07-02:
Check out the new version ... see if it works

superjasiek, 2020-07-02:
Thank you very much for reminder - i will try to test it this weekend 👍

closed 2022-01-13 by jameszah