jameszah/ESP32-CAM-Video-Recorder #12: FR: microphone, mqtt

M0ebiu5, 2020-07-19:
Great development so far James, thank you!
I have some ideas, maybe you like it:

- implement sound recording via mems microphone (i guess one has to decide between video or audio...)
-- trigger audio / video recording via noise level (maybe basic motion detection is also possible?)
-- a sync mechanism to start recording on multiple devices at once (so one can later mux the video and audio (see next point mqtt))

- mqtt interface (eg: request pictures as payload, change parameters, start / stop recordings, noise / motion alarm, etc)

jameszah, 2020-07-20:
Thanks.  I think I jammed as much functionality into the ESP32-CAM board as will fit, so I'm thinking of ordering some of the TTGO or ESP32-EYE boards -- the one with the microphone.  About 3x as expensive, but with the builtin usb, microphone, small display, .... and more gpio pins available.  I've spent a lot of time fighting with re-used pins 4, 12, and 13!

M0ebiu5, 2020-07-20:
I've ordered some i2s mems mics from ali - i hope the sensitivity is ok. The boards with display are pretty big...but if you get it working, i'm in.

As mentioned, it could be that audio works only without video (pins, ram, cpu...) with a reasonable effort, but for me it would be sufficient. And your code already provides most of the functionality...