jameszah/ESP32-CAM-Video-Recorder #14: Clean SD card automatically?

Ab-Tx, 2020-07-25:
Hi, does this project delete the older files of the SD card when there is almost no space available?
Also, How much time did you manage to record at what resolution and what size of SD card?
I'm planning on doing a surveillance camera with the Esp32-cam

jameszah, 2020-07-26:
It does not have an automatic delete function.  I've thought about doing it, but you can clean out old files with 2-3 clicks from your ftp client on your pc.  I've jammed a lot of stuff into this program, and a batch process running to maintain 20% free space could disrupt things.

The recording time is just a function of your choices.  
Reliable recording on a UHS-1 V10 card of:

2  fps UXGA quality 10 color
5  fps SVGA quality 10 color
10 fps VGA quality 10 color
20 fps CIF quality 10 color 

... all of those produce about 20 MB per minute.  Gray would be less.  So you can calculate 1 fps VGA etc if that is what you are planning.  And a 32 GB card works fine - I'm not sure if 64 GB works due to the format of the card.

Lately I've switched to lower quality (12) and the frame rate is a little more flexible, so I can get decent real-time-ish SVGA color video of 10 frames per second, at about 20 MB per minute.

Ab-Tx, 2020-07-26:
Nice, thanks a lot for the quick reply!

closed 2020-07-26 by Ab-Tx