jameszah/ESP32-CAM-Video-Recorder #15: parameters for short video

burkadius, 2020-07-31:
Thanks for this project ( v89 ). You can use it in many different ways as it seems to me. I have installed it and it works well. But I want to use it in such a way that I get an information plus image via "Telegram" when the PIR is activated and then only a short video of 1-2 minutes is recorded. Can you give me the example parameters with which I can achieve this goal.
Thanks
Burkadius

jameszah, 2020-08-01:
Generally you just set the framesize and the interval and quality, and put the PIRenabled at 1, and record_on_reboot to 0, and it should be all automatic.  That version waits for a PIR to go high, records a 15 second video according to your framesize/interval/quality and if there is still PIR activity in the last 5 seconds, it extends it for another 10 seconds.  So in a busy situation, the video could run for hours.

Your case of running for 2 minutes, then stopping, even with PIR activity, and then re-starting on the next PIR event ... would require a bit of a code change ... in a couple places.  Maybe I'll do that for a little experiment.  Stay tuned.

int record_on_reboot = 0;
int PIRpin = 13;                  
int PIRenabled = 1;               

int  framesize = 6;                
int  xspeed = 1;                   
int  gray = 0;                     
int  quality = 12;                
int  capture_interval = 100;       //  milli-seconds between frames

int Internet_Enabled = 1; 
int EnableBOT = 1;  ... with your telegram and bot numbers

burkadius, 2020-08-01:
Hi, thanks a lot. I will test it tomorrow.
Best Regards

closed 2021-01-21 by jameszah