jameszah/ESP32-CAM-Video-Recorder #16: Telegram Bot not working

Kobbe1, 2020-08-02:
Hi,

last version's re-write is really great, much cleaner with the separate settings. Great job!

For the Telegram bot - can't seem to get it to work. It says sucessfull in the serial monitor but nothing is received.
The code comments say "and then get your telegram number" - what number?

For BOTme - tried my phone number including the country code, tried my user name without the @, tried the bot's name.

Looked at https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot too ...

Pretty sure it is some dumb mistake :)

Best,

-T

Settings currently looks like this (not real credentials of course).
```
...
RTC_DATA_ATTR int EnableBOT = 1;
#define BOTtoken  "3xxx....xxb7"   // get your own bot and id at telegram.org 
#define BOTme     "33123468"
...
```

jameszah, 2020-08-02:
You need the 8 digit BOTme number (identity of your phone or browser at telegram), plus the 30-40 digit BOTtoken (identity of your esp32 at telegram).

The beginning of this article as a good explanation of getting those numbers:

https://randomnerdtutorials.com/telegram-control-esp32-esp8266-nodemcu-outputs/

Kobbe1, 2020-08-02:
Great - that worked! Thanks a lot.

closed 2020-08-02 by Kobbe1