jameszah/ESP32-CAM-Video-Recorder #17: OTA does not work

student1704, 2020-08-10:
Due to the fact that IO03 ist used for TX of the USB interface and PIR-output I tried to update the firmware via OTA, but there is no network connection shown in the Arduino IDE. The web-interface itself works properly.
Is there a workaround? 
What to do?

jameszah, 2020-08-10:
It is not that it doesn't work, it is that it is not implemented.

```
The is Arduino code, with standard setup for ESP32-CAM
    - Board ESP32 Wrover Module
    - Partition Scheme Huge APP (3MB **No OTA**)
```

To use OTA you have to partition the flash for the two full programs and add the OTA code.  It might be possible but all the ESP32-CAM programs specify Huge APP (no OTA), so it might be a challenge squeezing it all in.

You could re-use the RX and TX for other functions, but you would have to program the chip using RX and TX, then re-wire it to connect them to other things.  And re-do the wiring everytime you re-program the chip.

student1704, 2020-08-11:
Got it.
Thanks a lot for your support and the great app.

closed 2020-08-11 by student1704