jameszah/ESP32-CAM-Video-Recorder #21: Need assistance with settings.h

johnsonian01, 2020-08-26:
Hi, and thanks for keeping this project active. I'm unable to set the framesize , even though I modify settings.h as shown, it reverts to the value of 6, unless I use the WiFI connection to use your pre-assigned menus... then the framesize changes.
Any ideas? I'm pretty sure the sketch is using the settings.h located in the libraries folder (as it compiles) could there be a  copy in your 1.04 library?
Sam
nt  framesize = 7;                //  10 UXGA, 7 SVGA, 6 VGA, 5 CIF
int  repeat_config = 10;                 //  repaeat same movie this many times
int  xspeed = 1;

johnsonian01, 2020-08-26:
OK.. Getting this IDE figured out. Seems the IDE was opening the other libraries and including them in the tabs.. Changing text and using file/save did not seem to save the changes. When I use notepad, then the changes stick. Is this a correct understanding of the IDE?
Thanks:
Sam

closed 2020-08-26 by johnsonian01

jameszah, 2020-08-26:
I use the Arduino IDE with the 5-6 files open it tabs.  All the files should be in the local directory - everything referenced with the quotes

#include "settings.h"

... and the angle brackets are in the library directory ...

#include <EEPROM.h>

Another curve-ball is that it will read the settings from EPROM which will override the settings.h  It uses this line to determine if the settings in EPROM are good, so you could change the magic number 271 if you are recompiling with new setting.h and you want to skip the EPROM.

 if (ed.eprom_good == 271) {

johnsonian01, 2020-08-26:
AhHa... This makes sense, for sure...
Looks like with the right SD card, now I'm getting 10fps at SVGA.. so am happy: Thanks