jameszah/ESP32-CAM-Video-Recorder #22: SD_CARD Limitations

johnsonian01, 2020-08-27:
Hi All .. new to this forum and the ESP32.. Been making good progress looking at SD-CARDS, and indeed, , there can be a 4:1 difference in the achievable speeds with different grades and vendors. The best so far has been the SanDisk Ultra, rated 98MB/s, MicroSDHC UHS1... It labelled the same as the grey white version, which is 60% slower.(the fast one  has the in the grey/red package). Looks like the best I'm able to do is on a 800x600 video, 116527 bytes/frame in 264ms, so I'm averaging .35-.5 MB/s... This seems terribly below what the cards are spec'd.  Anyone know if a way to speed things up?
[image]

jameszah, 2020-08-29:
You can record about 10 fps svga on a UHS-1 circle10, but if you are also reading the card with the ftp interface, it will slow down.  

And the cards have all sorts of factors of reading/writing speeds, even within the V10 circle10 world.  I have not experimented with the V30, etc.  The code is not really optimized for super high speed, as it does several io's per frame, and keeps a running index in another file that only gets added at the end, so there are lots of small io's that will dramatically shrink the theoretical maximum of the sd card.

A faster solution would be to make a .mjpeg instead of an .avi, but the mjpeg would require more processing on the client side while you are displaying it, while the .avi is familiar to most devices.  There is another github that does that -- I think it is mentioned in the readme.

johnsonian01, 2020-09-02:
Yes, I am getting the expected fps now.

closed 2020-09-02 by johnsonian01