jameszah/ESP32-CAM-Video-Recorder #23: Turn on flash every capture

ricardo-brenner, 2020-08-27:
How can I enable the front led to each frame that is captured?

jameszah, 2020-08-28:
You want to turn on the Blinding Disk-Active Light?  It's blinding!

In function codeForCameraTask(), line 354 in v89, add the lines below.  The do_blink() function blinks it on slightly and briefly -- I'm not sure if that is currently used -- but if you want full power for a flash-photo, you can add these lines.

The picture is taken with the fb_get() -- you might have to delay for a couple milli-seconds for the led to turn on between the digitalwrite and the fb_get ???

https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/39b20d8963cc0c08c1d9be091e1f848260b89b15/v89/TimeLapseAvi89x.ino#L354

```
pinMode(4, OUTPUT);               // Blinding Disk-Active Light
digitalWrite(4, HIGH);             // turn ON

delay(1);  // delay to turn on the gpio, the transistor, and the LED ???

 bp = millis();
 fb_q[fb_in] = esp_camera_fb_get();
 totalp = totalp - bp + millis();

pinMode(4, OUTPUT);               // Blinding Disk-Active Light
digitalWrite(4, LOW);             // turn OFF

```

ricardo-brenner, 2020-08-28:
That's exactly what I needed, worked like a charm.

closed 2020-08-28 by ricardo-brenner