jameszah/ESP32-CAM-Video-Recorder #24: V89  - Time gap between  sending foto by BOT and starting AVI to long

student1704, 2020-08-28:
The time from sending BOT to starting avi is to long. How to overcome that?
During night Wifi is switched off and on in the morning controlled by internal time. At that time Internet_Enable is false.
 What happen with time during deepsleep because without Wifi NTP is not available.?

jameszah, 2020-08-29:
There is no clock on the esp32, so whenever you reboot or wake-up, it is Jan 1, 1970 and "In The Year 2525" is playing on Top 40 AM Radio ... and the ALOHAnet is not yet operational.   So don't sleep if you haven't got internet or a clock and you want to know the time.  😄 

It takes some time to deliver a picture to telegram.org -- it is a little finicky about getting bursts of messages.  You could send it while the avi is recording, but you will run out of heap, and the sd card will not work.  Or you could start deleting things like ftp, http, ... to find enough heap to run the camera, the sd card, and ssl at the same time.  I had it working once, but then I added something, and I could not record and telegram at the same time. ☹️

student1704, 2020-09-08:
got it. Thanks

closed 2020-09-08 by student1704