jameszah/ESP32-CAM-Video-Recorder #25: Compile error V94

johnsonian01, 2020-09-01:
Hi;;; Had V89 compiling perfectly, so setup a Sketch folder for V94 in the same directory as V89...:
Am getting the following compile error  ,as it seems a library issue:
Thanks:

TimeLapseAvi94x:881:33: error: 'httpd_req_t' was not declared in this scope
 static esp_err_t stream_handler(httpd_req_t *req) {
                                 ^
TimeLapseAvi94x:881:46: error: 'req' was not declared in this scope
 static esp_err_t stream_handler(httpd_req_t *req) {
                                              ^
[image]
[image]

jameszah, 2020-09-01:
Maybe check your libraries - I have the following: (below)

I think that httpd_req_t is in esp_http_server.h, and it was used in the previous version.  That declaration is for the new live stream and it is the first usage in the file.  

Check these lines are not scrambled somehow 

#include "esp_http_server.h"
#include <HTTPClient.h>

They are up near the top. 

And try a "Force-Recompile"  discussed here and elsewhere.
https://forum.arduino.cc/index.php?topic=215985.0

Might be simpler to put it in a new folder, and start fresh after restarting the IDE.
I notice there is an Arduino IDE update today -- which I have not done yet.

```
Using library ESP32 at version 1.0 in folder: C:\Users\James\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ESP32 
Using library EEPROM at version 1.0.3 in folder: C:\Users\James\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\EEPROM 
Using library WiFi at version 1.0 in folder: C:\Users\James\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi 
Using library WiFiClientSecure at version 1.0 in folder: C:\Users\James\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFiClientSecure 
Using library ArduinoJson at version 6.15.2 in folder: C:\Users\James\Documents\Arduino\libraries\ArduinoJson 
Using library ESPmDNS at version 1.0 in folder: C:\Users\James\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ESPmDNS 
Using library SD_MMC at version 1.0 in folder: C:\Users\James\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD_MMC 
Using library FS at version 1.0 in folder: C:\Users\James\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS 
Using library HTTPClient at version 1.2 in folder: C:\Users\James\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient
```

jameszah, 2020-09-01:
Third last point sounds ominous on the update!

```
ARDUINO 1.8.13 2020.06.16

[ide]
* Fixed crash on MacOSX >=10.15 with multiple monitor setups.
* The boards in the boards list in the "Tools" menu are now grouped by platform. This should make the selection
  easier when there are lots of boards loaded in the IDE. (thanks @matthijskooijman)
* Fixed save of proxy username/password settings in "manual proxy" section.
* In "Tools / Programmers" menu only shows programmers related to the currently selected board. (thanks @matthijskooijman)
* Fixed: sketch window out-of-display if the IDE is restarted after removing an external monitor.
* Allow baudrate/config change when the Serial Monitor/Plotter is disabled. (thanks @matthijskooijman)
* Updated jmdns to 3.5.5. Should improve network board detection.
* Optimized libraries scanning during startup. (thanks @ricardojlrufino)
* Better vertical centering of file names in tabs. (thanks @ricardojlrufino)
* Better console rendering of progress bar during upload. (thanks @matthijskooijman)
```

johnsonian01, 2020-09-01:
Changed local hard paths to <>,  and it seemed to help...
Now can not find the json file, but is is there... so getting closer:

ArduinoJson.h:5:29: fatal error: src/ArduinoJson.h: No such file or directory
compilation terminated.
exit status 1
src/ArduinoJson.h: No such file or directory

#include <esp_log.h>"
#include <esp_http_server.h>
#include <HTTPClient.h>
#include <esp_camera.h>

johnsonian01, 2020-09-01:
Yes... sure is a library thing, but Have been using Arduino 1.8.13 for the '89 build just fine. 
Still not able to find the arduinojson.h... had his problem with '89, so copied an instance into the sketch folder, and that helped,.. will keep trying.

johnsonian01, 2020-09-01:
Seems not to have the structure defined in the .ino...but it is defined in the esp_http_server.h just fine. and it can find the .h file... weird. Have deleted  the old directories, but now thinking it is not a library issue, unless I need to use library manager for more than just arduinojson   ...Am using the Wrover board, so maybe that's it?

Back to this.. The JSON worked when I deleted the library ArduinoJson version, and used library manger to re-install.

I have the esp_http_server.h in the sketch root, , and can see it is a structure, but is not declared in this scope...so it can fiind it, but needs to be place in scope somehow???

Multiple libraries were found for "WiFiClient.h"
 Used: C:\Users\Sam\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
'httpd_req_t' was not declared in this scope
[image]

johnsonian01, 2020-09-02:
This is the compile error related to the unsuccessful compile.... Within esp_http_server.h    the typedef for httpd_req_t   is defined, but the compiler seems to be throwing this in the .h file,  and possibly within the .ino also...
Wondering if others trying to compile version 94 also see this issue???

TimeLapseAvi94x:166:8: error: declaration does not declare anything [-fpermissive]
 static httpd_req_t;
        ^
C:\Users\Sam\Documents\ArduinoData\packages\esp32\hardware\esp32\ESP32-CAM-Video-Recorder-V94\TimeLapseAVI94x\TimeLapseAvi94x.ino: In function 'esp_err_t stream_handler(httpd_req_t*)':
TimeLapseAvi94x:885:49: error: 'esp_err_t stream_handler(httpd_req_t*)' redeclared as different kind of symbol
 static esp_err_t stream_handler(httpd_req_t *req) {
                                                 ^
C:\Users\Sam\Documents\ArduinoData\packages\esp32\hardware\esp32\ESP32-CAM-Video-Recorder-V94\TimeLapseAVI94x\TimeLapseAvi94x.ino:885:18: note: previous declaration 'esp_err_t stream_handler'
 static esp_err_t stream_handler(httpd_req_t *req) {

johnsonian01, 2020-09-02:
Am closing this out, as the root problem lies in a declaration of the http_req_t  type  in  esp_http_server.h...
Will keep trying to get V94 going..
Thanks for the useful info.

closed 2020-09-02 by johnsonian01

jameszah, 2020-09-02:
So I assume you got it working?

I tried new new Arduino 1.8.13, with the current Arduino esp32 Board Library 1.04 and the current ArduinoJSON 6.16.1 ... all set up as a portable installation so it has its own copy of all the libraries separate from your installed version of Arduino IDE ... and it compiles fine.

Here are the instructions for anyone interested:

1.  Install portable Arduino using Brian Lough instructions
https://www.youtube.com/watch?v=sTTY_w7Cuzk

2.  Make sure you create the /portable folder and move /arduino-1.8.13 up a level from the unzip folder C:\ArduinoPortable\arduino-1.8.13-windows

3.  Add board in Preferences
https://dl.espressif.com/dl/package_esp32_index.json

4.  Download board files for esp32 in Board Manager

5.  Download current ArduinoJSON (by Benoit) in Library Manager

6.  Download fresh copy of https://github.com/jameszah/ESP32-CAM-Video-Recorder
and put it in the /portable/sketch

7.  Unzip that, and copy the v94 folder up to /sketch and rename folder /TimeLapseAvi94x

8.  Preferences - turn on verbose for compile and download

9.  Open the TimeLapseAvi94x.ino, and compile, and it uses these libraries

```
Using library ESP32 at version 1.0 in folder: C:\ArduinoPortable\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\libraries\ESP32 
Using library EEPROM at version 1.0.3 in folder: C:\ArduinoPortable\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\libraries\EEPROM 
Using library WiFi at version 1.0 in folder: C:\ArduinoPortable\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi 
Using library WiFiClientSecure at version 1.0 in folder: C:\ArduinoPortable\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\libraries\WiFiClientSecure 
Using library ArduinoJson at version 6.16.1 in folder: C:\ArduinoPortable\arduino-1.8.13\portable\sketchbook\libraries\ArduinoJson 
Using library ESPmDNS at version 1.0 in folder: C:\ArduinoPortable\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\libraries\ESPmDNS 
Using library SD_MMC at version 1.0 in folder: C:\ArduinoPortable\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\libraries\SD_MMC 
Using library FS at version 1.0 in folder: C:\ArduinoPortable\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\libraries\FS 
Using library HTTPClient at version 1.2 in folder: C:\ArduinoPortable\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient 
"C:\\ArduinoPortable\\arduino-1.8.13\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-size" -A "C:\\Users\\James\\AppData\\Local\\Temp\\arduino_build_237202/TimeLapseAvi94x.ino.elf"
Sketch uses 1158062 bytes (36%) of program storage space. Maximum is 3145728 bytes.
Global variables use 54416 bytes (16%) of dynamic memory, leaving 273264 bytes for local variables. Maximum is 327680 bytes.

```

johnsonian01, 2020-09-02:
Awesome... let me try this approach.. Thanks!