jameszah/ESP32-CAM-Video-Recorder #26: Can´t Upload

Ansmanner, 2020-09-06:
At first nice work. Ive seen the result in a youtube-video.
Now to my problem:
I connectet the ESP-32 Camera Module properly via an Arduino to the Computer. Then I selectet the option as said in the Script:

 *  Board Settings:
 *  Board: "ESP32 Wrover Module"
 *  Upload Speed: "921600"
 *  Flash Frequency: "80MHz"
 *  Flash Mode: "QIO"
 *  Partition Scheme: "Hue APP (3MB No OTA/1MB SPIFFS)"
 *  Core Debug Level: "None"
 *  COM Port: Depends *On Your System*

Now when I try to upload the Sketch the following error shows up:
A fatal error occurred: Timed out waiting for packet header
`
esptool.py v2.6
Serial port COM4
Connecting........_____.
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: _________________
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header

`
The sample Sketch worked.
Or is there a instruction to upload it properly?

Maybe you have a solution.
Thank you!

johnsonian01, 2020-09-06:
I have the same issues a lot when there is a contention between another app trying to use the COM port(like Realterm, Putty). I just close them all down, remove and replace the UART dongle, and usually, I upload. the ......______ that you see is exactly what I get. MAybe this can Help.

johnsonian01, 2020-09-06:
Just noticed. I use 115200, so not sure how the MAC works this out.

jameszah, 2020-09-06:
1. Make sure you have the correct COM port, and then open a Tools - Serial Monitor

2. Pin GPIO0 (gpio zero marked IO0) to GND.

3. Cycle the power on the chip so it starts with gpio0 grounded, and the serial monitor should say:

```
13:02:46.075 -> ets Jun  8 2016 00:22:57
13:02:46.075 -> 
13:02:46.075 -> rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
13:02:46.075 -> waiting for download
```

... and then you are ready to download.

After download, Unpin the gpio0 from ground, cycle the power again, and it should boot.

jameszah, 2020-09-06:
And make sure the rx and tx are correctly aligned, and power is 3.3v.  
I think there are lots of explanations of that.  I did it once or twice before getting an ftdi.

If you see that message "waiting for download" then you should be set.

Ansmanner, 2020-09-08:
Thanks now I was able to flash the version 39.
Now I get some weird lines in the Serial Monitor:

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x2fbf0004,len:192
ho 0 tail 12 room 4
load:0x50571055,len:357568597
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
1162 mmu set 00060000, pos 00060000
1162 mmu set 00070000, pos 00070000
1162 mmu set 00080000, pos 00080000
1162 mmu set 00090000, pos 00090000
1162 mmu set 000a0000, pos 000a0000
1162 mmu set 000b0000, pos 000b0000

and so on.
I can see the module in the router with the IP 192.168.112.33

I edited the IP stuff in the code: 

 //  this is the fixed ip stuff that does not work with with router
    // zzz
    // Set your Static IP address
    IPAddress local_IP(192, 168, 112, 33);
    //IPAddress local_IP(192, 169, 1, 225);

    // Set your Gateway IP address
    IPAddress gateway(192, 168, 112, 1);
    //IPAddress gateway(192, 169, 1, 1);

    IPAddress subnet(255, 255, 255, 0);
    IPAddress primaryDNS(8, 8, 8, 8); // optional
    IPAddress secondaryDNS(8, 8, 4, 4); // optional

    WiFi.mode(WIFI_STA);

    WiFi.setHostname("ESP32CAM225");  // does not seem to do anything with my wifi router ???

    if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
    Serial.println("STA Failed to configure");
    major_fail();
    }

What am I doing wrong now?
Thank your for your fast replys

Ansmanner, 2020-09-08:
Also the sd card is empty when I plug it into the computer
...

jameszah, 2020-09-09:
I have not seen those "1162 mnu ..." messages - maybe something to do with writing the flash?  The first 8 lines look normal.  So there is no text after the boot - no debug messages?

But if it boots okay, you can ignore that.  You say it shows up on the network -- did you click on that ip address -- you should see the recording status and a picture through the camera.

The different versions have different default parameters.  Look for the blinking red led on the back.  Does it blick SOS - 3 long - 3 short - 3 long -- or is it blinking at a regular pace -- which would mean it is recording.

If you click on the ip adress, you can stop the recording, and move to the sd to your pc.

~~~~~~~~~~~
A quick google of "1162 mnu" gets this

https://github.com/espressif/arduino-esp32/issues/58

This "igrr" guy is the expert of esp32 -- he seems to think it is power related.  So you could check you have good thick wires to your +5 and GND.

seeman78, 2021-12-30:
i have this error, how can i fix it?

C:\Users\xxxx\Documents\Arduino\TelVideo\ESP32-CAM-Video-Telegram\ESP32-CAM-Video-Telegram.ino: In function 'void send_the_video()':
ESP32-CAM-Video-Telegram:1152:22: error: 'class UniversalTelegramBot' has no member named 'sendMultipartFormDataToTelegramWithCaption'
   String sent2 = bot.sendMultipartFormDataToTelegramWithCaption("sendDocument", "document", strftime_buf,
                      ^
ESP32-CAM-Video-Telegram:1153:51: error: 'chat_id' was not declared in this scope
                  "image/jpeg", "Intruder alert!", chat_id, psram_avi_ptr - psram_avi_buf,

seeman78, 2021-12-30:
or can you help witch Programm code?
[image]

jameszah, 2021-12-30:
Do you have all these files in the same folder?
[image]

[image]

jameszah, 2021-12-30:
... or the similar files for the video telegram program

seeman78, 2021-12-31:
[image]
[image]
i understand. but now its hang on by PIR Setup, can i use it without pir?

seeman78, 2021-12-31:
oh, now its super, thank you

closed 2022-01-12 by jameszah