jameszah/ESP32-CAM-Video-Recorder #27: How to do a forced-recompile (again)

johnsonian01, 2020-09-06:
In working with V94, I have been successful in changing the settings.h file back and forth many times, since I I'm leaving the WiFi disabled due to the distances I am flying.  This has worked for the last few days... Now, I am stuck in the UXGA mode, while the settings.h has been changes to SVGA... It is like the compiler is again using cached older .o files or the like, and not updating with the new info. I believe what I want to do is a "forced Recompile"... I have been trying the ideas put forth in the closed ticket V94 compile problems(now closed) , including temporarily changing the board info (nothing), putting the project in a new portable folder, and changing the preferences to set the project up there( nothing).  I have set the entire structure up as portable, as recommended, and using the arduino.exe located in the location C:\ArduinoPortable\arduino-1.8.13-esp32, as this has worked before. 

Any Ideas how to finally get past this problem?
Thansk

johnsonian01, 2020-09-06:
Last thing James told me was that the MagicNumber in settings (was 314).. I only changed it by one  to 313, and still all the settings were copied over from the EEPROM. Changing the MagicNumber ( to 100) now disables the EEPROM, and the settings now load from settings.h.
Problem solved. Brain Fart & Senior Moment.

johnsonian01, 2020-09-06:
Hmmmm.. on second thought, found out something interesting. Could the EEPROM be storing the most recent downloaded MagicNumber,, and if it doesn't change on the  next download, the EEPROM values are loaded instead of the settings.h..
Sure seems that way, for if I increment or change each magicnumber each successive downloads, the settings.h values always work!!!

jameszah, 2020-09-07:
Ahhh ... it is supposed to erase the EPROM and use the settings.h on the first time after a new compile, if you changed the MagicNumber.  After that it writes the MagicNumber into the EPROM with the current settings, and will re-use those every reboot because the EPROM and compiled MagicNumber will be the same.  It will also update the EPROM (with the compiled MagicNumber) every time you do a /start.  If you recompile but don't change the MagicNumber, then EPROM and Compile are the same, so it can use the EPROM settings.  

It is just a test to see that the EPROM has valid info, and the format has not changed since it was last written.

Most of the parameters can be changed with the /start command, so this doesn't matter.  But "InternetEnabled" cannot be changed with /start, so that comes from the settings.h and then written into EPROM, from where it will be fetched on all subsequent reboots until the next compile. 

I think if it changed 314 -> 313 the "equal" test should fail and it should use settings.h.   I think it is an integer.

The issue of booting with Internet, then leaving the Internet zone while camera is still running, ... I think should be okay.  The internet stuff will all fail, and it will try to reset/restart internet from time-to-time, but that should be okay I think.  Just some  unnecessary cpu time.

closed 2022-01-13 by jameszah