jameszah/ESP32-CAM-Video-Recorder #28: No download possible with Android FTP client

student1704, 2020-09-08:
I like to work with V89. But I can not download the videos with Android FTP client. It seems , Login is ok, but LIST  or LISTchild can not deal with the file name or file date....
FileZilla with Win10 works properly. 
Any idea is appreciated

jameszah, 2020-09-08:
Could you give me a specific name for that Android FTP client, and I'll have a look at it.
There are a bunch of them.

student1704, 2020-09-08:
I tried:
Dateimanager 2.5  (Filemanager)   but this works fine with ESP32 standard version and lib "ESP32_FTPClient"
AndFTP  
Total Commander 
All of them do not show the contents of root or child dirs.

jameszah, 2020-09-09:
The ftp server I am using does not do all ftp commands and some clients try to use all the features.

I'll have a look.  ...

When you say  "ESP32 standard version and lib "ESP32_FTPClient" -- is that an ftp client on a esp that is accessing the ftp server on the different esp32-cam esp?

jameszah, 2020-09-10:
I tried 5-6 android (samsung phone) ftp's and could not find anything good.  AndFTP seems to the the best - it connects properly, and you can send files to the ESP-CAM, and create directories, and there is a little window at the bottom where you can see the commands rolling past showing that it is getting the listing of files, but it does not show anything on the screen, so you cannot click and download them.  You can set the directory on the config - like 20200909 - and it will show that there are 27 files, but not display them.  So file-length? or format (underscores? spaces?)  or something. 

No solution so far.

closed 2022-01-13 by jameszah