jameszah/ESP32-CAM-Video-Recorder #29: Is it possible to record on the SD and then send to Telegram?

Heist87, 2020-09-19:
Is it possible to record on the SD and then send to Telegram?  Please I need that, now i have only photos to Telegram but I would want videos

jameszah, 2020-09-21:
Possible - but too much in here already.  Maybe I'll combine:

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior

and

https://github.com/jameszah/ESP32-Cam-Telegram

But the video's are very big mjpeg avi files ... and telegram is a little sensitive to getting loads of data.

I could try a 10 seconds vga video ... for an example.  Stay tuned.

Heist87, 2020-09-21:
> Possible - but too much in here already. Maybe I'll combine:
>
> https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior
>
> and
>
> https://github.com/jameszah/ESP32-Cam-Telegram
>
> But the video's are [...]

I will wait patiently because I want that even if it is 10 seconds, I will have enough with that, thank you very much

espcam, 2021-04-29:
this would be awesome feature, video quality does not matter so much, 5-10 seconds is good enough to check what happens

jameszah, 2021-05-01:
How would that work?  This is basically a record-to-sd program, and has the photo-to-telegram at beginning of movie, and has the web-photo or web-streaming on demand.  So the video-to-telegram would be used on demand?, or at regular intervals?  I've thought of getting rid of the sd on a new program, and just have web-stream-on-demand, web-photo-on-demand, and short video-to-telegram on event, like a pir trigger, or a click on a doorbell.  

I suppose if you only record-on-pir, then video-to-telegram and video-to-sd at the same time would be logical.

closed 2022-01-13 by jameszah