jameszah/ESP32-CAM-Video-Recorder #31: How to set default recording settings? and override oldest file when SD is full?

alexw22, 2020-09-26:
First off, great project!! I have been using it for some time already and it is absolutely amazing! I am always eager to immediately try out new versions you post :-)
 
How do I set the default recording settings? I mean the initial settings after programming the board? Up to now I have been going into the web interface to change the settings to fit my needs.

And would it be possible to implement a feature so that the ESP would override and replace the oldest recording if the SD card is full?

jameszah, 2020-09-27:
re: settings
Assuming you are using the latest edition -- it first looks in the EPROM and if the MagicNumber in EPROM matches the code, then it will use the EPROM settings.  If it does not match, it will use the settings.h settings.  Either way, it will then write the settings and the MagicNumber into the EPROM to use on the next boot.  Then wherever you sent a /start command, it will write those settings into EPROM for the next boot.  Some settings, like disable the internet, or record-on-reboot cannot be changed with the /start command, so they have to be changed in settings.h and re-downloaded to the esp32.  You just have to change the MagicNumber to any other value, to tell the program to ignore the EPROM and use the settings.h.

regarding deleting the old files -- frequent request!  You can do that quickly with the ftp system, but and automatic delete would be nice.  I have it on my list.  Maybe I'll get to it. 😄

alexw22, 2020-09-29:
Changing the MagicNumber indeed looked like it changed the starting values however I am getting some unexpected behaviour. Perhaps I am misunderstanding the variables in settings.h.

My settings from the settings.h file:
int  framesize = 7;                //  10 UXGA, 7 SVGA, 6 VGA, 5 CIF
int  repeat_config = 1;                 //  repaeat same movie this many times
int  xspeed = 3;                   //  playback speed - realtime is 1, or 300 means playpack 30 fps of frames at 10 second per frame ( 30 fps / 0.1 fps ) 
int  gray = 0;                     //  not gray
int  quality = 12;                 //  quality on the 10..50 subscale - 10 is good, 20 is grainy and smaller files, 12 is better in bright sunshine due to clipping
int  capture_interval = 100;       //  milli-seconds between frames
volatile int  total_frames_config = 3000;  //  how many frames - length of movie in ms is total_frames x capture_interval

My aim is to record for 5 min at 10fps. When the PIR sensor is activated it makes one file that will be played back at 3x speed. When going into the web interface I see that Length = 15 seconds?? I am expecting this to be 300 seconds or 5 min (or 100 seconds if it is counting 3x playback speed)
[image]

Slightly confusing for me is:
1. the difference between "int  framesize = 7;" and "int  quality = 12;":
But I am ok leaving them as they are for now

2. The meaning of "int  repeat_config = 1;":
In the example sketch the values are pretty high! But I am expecting that was initially used for when the ESP would not be retriggered by a PIR so that it would keep recording for some time and make a couple files just in case one of the files ends up being corrupt or something. I just want to create one file per PIR activation so the best I could come up with was to set this value to "1"

3. "int  capture_interval = 100;" and "volatile int  total_frames_config = 3000;"
10fps and I want it to continue for 5 min after which it would idle until the PIR would be reactivated. So if my calculations are right that would make 10/1sec -> 600/1min -> 3000/5min

alexw22, 2020-09-29:
Got it working after another upload. Sorry for the confusion. Not sure what went wrong the first time. Now length is displayed as expected (300 sec).

jameszah, 2020-09-30:
ok, you are the beta-tester -- this should delete a day of old videos whenever your SD is 90% full, and it does it before each video start -- debug messages on serial.

https://github.com/jameszah/ESP32-CAM-Video-Recorder/tree/master/v98

There are also a bunch of compile directives to delete ftp, telegram, etc ... in settings.h

alexw22, 2020-09-30:
Fantastic!! I’ll give it a try as soon as I get back home next week as I am out of the workshop until then! I’ll get back to you in case I find any bugs or unexpected behaviour :-)

closed 2021-01-21 by jameszah