jameszah/ESP32-CAM-Video-Recorder #32: PIR sensor documentation clarification

alexw22, 2020-10-27:
When I first added a pir sensor to the camera I left the pir sensor to its default settings that it came with from the mail. A couple questions that came up while reviewing my pir sensor settings as I noticed my video files are shorter than expected which could be due to my pir settings I discovered. Anyhow a picture and a sentence or two on the main page would be valuable for others replicating this project:

Most of the pir sensors have 3 adjustable settings. Sensitivity adjustment potentiometer, Triggered time adjustment potentiometer and Retriggerable and non-retriggerable jumper switch or solder pads depending on the variation you have.

[image]

1. Sensitivity is the amount of heat sensed that can trigger the sensor. Is this image is anything to go by clockwise rotation would make it more sensitive and counter clockwise less sensitive:
[image]

2. Trigger time adjustment or adjust delay time as they name it in the picture above is the amount of time the signal stays high when the sensor is triggered. Clockwise rotation would make the output pin of the sensor stay high for a longer time when the sensor is triggered and counter clockwise rotation would make the high output last shorter after triggering.

3. The retriggerable and non-retriggerable jumper switch or solder pads, when set to the “H” setting it means that it is retriggerable and that if the sensor is triggered and after a couple seconds when the output is still high it would restart the trigger time adjustment so that the high output signal is maintained for a longer time. If the sensor would be retriggered constantly it would never go low theoretically. If the jumper is set to the “L” setting which is non-retriggerable. It means that when the sensor is triggered and the output goes high the sensor can not be retriggered to for the duration for the time delay even when a heat source is moving I front of the sensor and the output fill go low after the time delay set with the potentiometer will run out and the sensor output will go low before it goes high again if the heat source is still moving in front of the sensor.

In my case I was experiencing that most videos on my SD card were shorter than expected and that some videos were longer. This could be due to two things that I am currently troubleshooting. First it could be due to that my 64Gb SD card that I formatted to FAT32 is not working well. I switched it to another 32Gb SD card now. While changing the SD card I thought it might be also due to my PIR as it was set to the “L” setting with the jumper. So set to non-retriggerable. So this means that if the motion is detected for a longer time than my 
trigger time adjustment was set that the output of the sensor would go high, low, high, low. Etc. Is the camera starting a new recording every time it goes high after being low? This could have been why some of my videos on the SD card were shorter than expected.

I hope you will find the information above useful and saves you at least the time to type it out yourself on the information/main page of this awesome GitHub project!

Meins321, 2022-08-03:
well it would make sense to make the record time set by a given number in the website served...

the pir sensor might jumpy or just provide a very long high signal much longer than the record time wanted..

How is this determined in software?

jameszah, 2022-08-04:
https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/b6c9af8970bc2321935662286603da7ce6484271/vA1/TimeLapseAviA1x.ino#L540

This one records 15 seconds after the signal goes off - so 15 sec minimum - but will record for as long as the pir is showing active.  If it is bouncy within the 15 seconds, you just get the one video.  So it works with a pir that has hardware to keep it on for a certain duration, or a pir that just gives a pulse.

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior

This other program just records when the pir is active by connecting the pir to the "enable record" pin -- it is shown as an on/off switch to control the recording with a switch, but the pir can control that switch too.