jameszah/ESP32-CAM-Video-Recorder #33: Is it possible to include the field input for a user telegram token,wifi name&password? on a control web page interface,for flexibility to users?

dollatron, 2020-11-01:


jameszah, 2020-11-01:
I've got WiFiManager configuration on my todo list.
Originally I was going to set up camera with it, but I did that with the /start calls, but still needs the ssid and telegram config ...
Working on a few other things first. 😄

dollatron, 2020-11-07:
> 
> 
> I've got WiFiManager configuration on my todo list.
> Originally I was going to set up camera with it, but I did that with the /start calls, but still needs the ssid and telegram config ...
> Working on a few other things first. 😄

ok that will be superb, i have been trying to make it so .

toome123, 2020-12-07:
Here is some library that looks nice and I think will do the job.. [https://github.com/Hieromon/AutoConnect](https://github.com/Hieromon/AutoConnect)

dollatron, 2021-02-09:
> 
> 
> Here is some library that looks nice and I think will do the job.. https://github.com/Hieromon/AutoConnect

thanks mate

closed 2021-02-12 by jameszah