jameszah/ESP32-CAM-Video-Recorder #35: avoiding hardcoding wifi credentials by using wifi  a manager

dollatron, 2020-11-29:
i found this link useful(https://youtu.be/xO6SISq-DQ4) for wifi manager where you can connect to any wifi hotspot if you know the password..but i need some help to intergate those codes to 
ESP32-CAM-Video-Recorder

jameszah, 2020-11-29:
I gave it shot, in a limited window of time today, and came up with these problems:

1.  WiFiManager by tzapu, which I have used before on 8266, does not have an official esp32 version
2.  many WiFiManagers clones of tzapu for esp32 version, ... ???
3.  Tried IotWebConf - which works nicely on esp32 in the example program, plus has "other" parameters, for config parameters
4.  IotWebConf uses same section of eprom as my code, so I had to move one (not too hard)
5.  IotWebConf and my code (through esp http server) use different version of web servers libraries, so I have to redo the IOTWebConf with the fancier library (HTTPMethod' to 'httpd_method_t {aka http_method)  ...

But I did learn a lot of this code, so I can come back to it later.
I also learned that the esp32 does not have any eprom, but just uses flash to implement a fake eprom!  Who knew?

dollatron, 2020-12-05:
> I gave it shot, in a limited window of time today, and came up with these problems:
>
>     1. WiFiManager by tzapu, which I have used before on 8266, does not have an official esp32 version
>
>     2. [...]

thanks for your contribution on this,i was getting lots of compilation errors concering (HTTPMethod' to 'httpd_method_t {aka http_method) )i end up thinking wifiManager library for ESP32 needs som compatibility issues..let me explore what you have found out about IotWebConf

toome123, 2020-12-07:
Here is some library that looks nice and I think will do the job.. [https://github.com/Hieromon/AutoConnect](https://github.com/Hieromon/AutoConnect)

jameszah, 2020-12-09:
Made a little bit of progress on this issue.

There seems to be a feud between the include file esp_http_server.h which calls the esp32 arduino core http functions, which I am using to implement the web server, and the include file WebServer.h which this IotWebConf system is using, which is a library built on top of the esp32 arduino core functions which implements the object WebServer used by IotWebConf -- and as far as I can see all of the these WiFiManager-like applications.

But esp_http_server.h includes http_parser.h which has a complex include macro which defines among other things "HTTP_GET     = 0b00000001".  Alas WebServer.h includes HTTP_Method.h which also defines HTTP_GET and that spews out dozens of errors and leaves this unforgivable error  

`cannot convert 'HTTPMethod' to 'httpd_method_t {aka http_method}`

This is used by my code:

```
//#define jz_HTTP_GET   (httpd_method_t)0b00000001
  httpd_uri_t index_uri = {
    .uri       = "/",
    //.method    = jz_HTTP_GET,
    .method    = HTTP_GET,
    .handler   = index_handler,
    .user_ctx  = NULL
  };
```

... and my first try was to dodge the problem by NOT using the HTTP_GET symbol (using instead jz_HTTP_GET), and let WebServer.h own that symbol.  That did not work ... cannot remember why just now (?)

So then I let esp_http_server.h and http_parser.h own the symbol and take it away from WebServer.h and HTTP_Method.h -- this did work, and you just have to include the following code in the main program:

```
#include "esp_http_server.h"  // v99
#define _HTTP_Method_H_ // v99

typedef enum {
  jHTTP_GET     = 0b00000001,
  jHTTP_POST    = 0b00000010,
  jHTTP_DELETE  = 0b00000100,
  jHTTP_PUT     = 0b00001000,
  jHTTP_PATCH   = 0b00010000,
  jHTTP_HEAD    = 0b00100000,
  jHTTP_OPTIONS = 0b01000000,
  jHTTP_ANY     = 0b01111111,
} HTTPMethod;

// iotWebConf stuff v99
#include <IotWebConf.h>
const char thingName[] = "jz";
const char wifiInitialApPassword[] = "jz12345678";
DNSServer dnsServer;
WebServer server(80);

IotWebConf iotWebConf(thingName, &dnsServer, &server, wifiInitialApPassword);
```

The _HTTP_Method_H_ define prevents HTTP_Method.h from trying to redefine the symbol HTTP_GET as "1", and renaming the HTTPMethod type enum components with the letter "j" does not cause any trouble with the WebServer.h library.  So now it all compiles and links without any problems and it basically works.

BUT ... the program does run, and after IotWebConf reads an old configuration from eprom/flash and then starts wifi properly, and my code of using  esp_http_server.h starts its web indentity which in turn runs the web server, plus the ftp server on top of the wifi server, plus wificlientsercure which is required for the telegram library, .... and everything is fine ....

BUT ... if I let IotWebConf starts WebServer.h in the normal way to let you set the name and password of your wifi, then all of my code with its own webserver, ftp, telegram, ... will not start, as the WiFi, tcpip, Web, etc belongs to WebServer.h and esp_http_server.h cannot use it.  So calling http://192.168.172/" returns "Not found: /" , so everything has to be run through WebServer.h constructs.  My code returns the fatal error, which means none of my web stuff works.

`E (39095) httpd: httpd_server_init: error in bind (112)
`

So the WebServer.h library has grabbed control, and esp_http_server.h cannot get started.  If you start esp_http_server.h first, it will start fine and do its functions, but I'm guessing WebServer.h will fail to start as a network, but it will still function to read details from flash/eprom.  

So maybe the solution is to start WebServer.h and set the wifi parameters, then reboot, and not start WebServer.h and just start esp_http_server.h with all the network stuff I was using.

I fear this might fall under the dominion of @igrr and @me-no-dev.

Relevant links:

https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/src/WebServer.h
https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/src/HTTP_Method.h

```
#ifndef _HTTP_Method_H_
#define _HTTP_Method_H_

typedef enum {
  HTTP_GET     = 0b00000001,
  HTTP_POST    = 0b00000010,
  HTTP_DELETE  = 0b00000100,
  HTTP_PUT     = 0b00001000,
  HTTP_PATCH   = 0b00010000,
  HTTP_HEAD    = 0b00100000,
  HTTP_OPTIONS = 0b01000000,
  HTTP_ANY     = 0b01111111,
} HTTPMethod;

#endif /* _HTTP_Method_H_ */
```

https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/esp_http_server/esp_http_server.h
https://github.com/espressif/esp-idf/blob/178b122/components/esp_http_server/include/esp_http_server.h
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/nghttp/http_parser.h

```
enum http_method
  {
#define XX(num, name, string) HTTP_##name = num,
  HTTP_METHOD_MAP(XX)
#undef XX
  };
```

jameszah, 2020-12-10:
So I got this working ... in a very unsatisfactory manner.

Basically you let iotWebConf control the web traffic for 1 minute during the setup with the call iotWebConf.init and repeated calls to iotWedConf.doLoop .... during which it connects to the 192.168.4.1 to let you configure your wifi name/pass and other parameters, and then if connects to that new wifi, or if justs connects to that wifi that you previously configured.

And then make the call server.stop() which will shut off the WebServer object, and the I can start the other system of web server with the calls httpd_start(&camera_httpd, &config), httpd_register_uri_handler(camera_httpd, &index_uri), httpd_resp_send, httpd_resp_send_chunk, ... that runs the web server in this program.

And so the IotWebConf.doLoop is not called during the loop(), and you cannot change you web configuration during the loop(), just after the reboot while setup() is running.

A nice solution would be to redo the web-calls in IotWebConf do drop the WebServer object, and switch to the httpd_ style of calls, and httpd_ can handle the web control of the camera and streaming, and it can also trigger IotWebConf calls to set it up at the beginning, and let you edit your parameters over the web while the esp32 is doing its regular loop() activities.  I don't think it would be too hard, but httpd_ is a lower level fucntion calls the the WebServer stuff. 

And it would settle the feud between WebServer.h and esp_http_server.h about who owns http_method and HTTP_GET symbol.
```
#include "esp_http_server.h"  // v99
#define _HTTP_Method_H_ // v99

typedef enum {
  jHTTP_GET     = 0b00000001,
  jHTTP_POST    = 0b00000010,
  jHTTP_DELETE  = 0b00000100,
  jHTTP_PUT     = 0b00001000,
  jHTTP_PATCH   = 0b00010000,
  jHTTP_HEAD    = 0b00100000,
  jHTTP_OPTIONS = 0b01000000,
  jHTTP_ANY     = 0b01111111,
} HTTPMethod;

// iotWebConf stuff v99
#include <IotWebConf.h>
# define IOTWEBCONF_CONFIG_START 100
const char thingName[] = "jz";
const char wifiInitialApPassword[] = "jz12345678";
DNSServer dnsServer;
WebServer server(80);

IotWebConf iotWebConf(thingName, &dnsServer, &server, wifiInitialApPassword);
```

```
void setup() {

    Serial.println("Let IotWebConf run for 1 minute");

    iotWebConf.init();
    server.on("/wifi", []{ iotWebConf.handleConfig(); });
  
    long wifistart = millis();
  
    while (millis() - wifistart < 60000){
    
        iotWebConf.doLoop();
        delay(100);
    }
   
    server.stop();
    Serial.println(" IotWebConf done ");

   // hereafter assume iotWebConf has started wifi, and do your own networking stuff
  // such as time, mdns, ftp over wifi, httpd_start, UniversalTelegramBot bot(BOTtoken, client); over WiFIClientSecure, ...

}  // end of setup
```
  

And nothing from IotWebConf in the loop().

jameszah, 2021-01-06:
Got the WiFiManager working.  Check it out.

closed 2021-01-21 by jameszah

dollatron, 2021-02-09:
Hello friend, i really appreciate your effort ..am going practically now
using your last update.

dollatron, 2021-02-09:
very unsatisfactory manner but am super excited to learn alot about this
project

dollatron, 2021-02-09:
super appreciated💪

dollatron, 2021-02-09:
💯💯

ashish-iottive, 2021-06-30:
Hey @jameszah Is there any code where [WiFi Manager](https://github.com/tzapu/WiFiManager) Library works with only ESP32-Camera Webserver code cause I need a code integrated with Arduino ESP32 example Camera Webserver with WiFi Manager.

jameszah, 2021-06-30:
Here is a summary of the changes to use the WebServer.h (c++ object) and esp_http_server.h (regular c functions) at the same time.

https://github.com/tzapu/WiFiManager/issues/1184#issuecomment-754255856

I don't recall if the Camera Webserver example uses the object, but connecting to the previous AP might just be the AP stored in the the ESP32 wifi system somewhere -- you can clean that out with 

`    WiFi.disconnect(true, true);`

which should clean the previous connection, and use the WiFiManager or hardcoded parameters rather than the last good connection.