jameszah/ESP32-CAM-Video-Recorder #36: UniversalTelegramBot.cpp:132:37: error: 'measureJson' was not declared in this scope

fidoboulettes, 2020-12-30:
Hi Jamesah,
sorry for my poor english, i don't speak it very well.
First, thanks for your great job!!!

I'm facing an issue, when I test your code, Arduino IDE respond this message:

_sketch\UniversalTelegramBot.cpp: In member function 'String UniversalTelegramBot::sendPostToTelegram(String, ArduinoJson::JsonObject)':
UniversalTelegramBot.cpp:132:37: error: 'measureJson' was not declared in this scope
     int length = measureJson(payload);

I don't understand the problem, can you help me?

Regards,

jameszah, 2020-12-30:
Do you have the ArduinoJson library installed?

Also check the "#define include_telegram" is in settings.h, which controls this little block in the manin program, which includes all the telegram stuff

```
#ifdef include_telegram
#include <WiFiClientSecure.h>
#include "UniversalTelegramBot.h"
WiFiClientSecure client;
UniversalTelegramBot bot(BOTtoken, client);
#endif
```

Here is how to add/check the json library:

In Arduino, go to Tools -> Manage Libraries -> search for json, and install the highest release of version 6 by Benoit.

Using library ArduinoJson at version 6.15.2 in folder: C:\Users\James\Documents\Arduino\libraries\ArduinoJson
  
[image]

fidoboulettes, 2020-12-30:
Ok Jamesah,
The update of Arduinojson lib (6.15.1 --> 6.17.1) works.. Thanks.

I have a new problem... The cam can't mount my sdcard (8GB and 32GB)... I've tried to format in fat 32 and Exfat but same problem...
Have you an idea?

Thanks a lot

jameszah, 2020-12-31:
Excellent - my advice would have been fresh fat32 format -- and make sure pin 12 and 13 are floating.  12 and 13 and are connected to the SD and sometimes other hardware, and can cause problems.

closed 2021-01-21 by jameszah