jameszah/ESP32-CAM-Video-Recorder #38: Does not record after disabling telegram

huseman21, 2021-01-27:
If I comment out   //#define include_telegram inside settings.h   it never writes to the sd card again.

I can start a new recording,  the serial output says its recording but its

zzzstarting recording
zWrite settings in the EPROM 
Writing to EPROM ...
Put took 8 ms, bytes = 60
do_start Starting a new AVI
zzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzz

jameszah, 2021-01-27:
check EnableBOT == 1 in the settings.h file (near the bottom).  It should be "0" or not enabled.

The "#define include_telegram" will have cut out the code to send the telegram, but the Enable_BOT of 1, will still try to send the telegram.

And since the Enable_BOT is in EPROM, which will override the setting in setting.h, then you have to go into settings.h and change MagicNumber to something else, so the EPROM settings are ignored, and the setting in settings.h of Enable_BOT = 0 is used.

I suppose removing the telegram code should set Enable_BOT to 0 automatically to avoid this problem.

jameszah, 2021-01-27:
... or you could just disable it from the http interface ... that would be quicker

http://desklens.local/start?bot=0

huseman21, 2021-02-03:
ok ill try that.

closed 2021-02-03 by huseman21