jameszah/ESP32-CAM-Video-Recorder #40: get configured fps api on web server

quockhanh1198, 2021-02-22:
Hi @jameszah , how can I get configured fps api by http_get when I set in arduino code ?, thank you so much !

jameszah, 2021-02-28:
You want a particular frames-per-second to be sent to your computer web browser?

In settings.h there is the line

`int stream_interval = 333;         // milliseconds between frames delivered during the live stream - 333 is 3 fps`

You can modify that to 200 for 5 fps, etc.  It can use up resources that were going to recording, but you can experiment with it.

quockhanh1198, 2021-02-28:
Thanks for your reply, so how can I only stream on browser without store video or record video from SD card, bro ?

jameszah, 2021-02-28:
If you just want streaming, you can use the free program that comes with the esp32 arduino library.

[image]

Here is a good introduction video and website:

https://randomnerdtutorials.com/esp32-cam-video-streaming-web-server-camera-home-assistant/

closed 2022-01-13 by jameszah