jameszah/ESP32-CAM-Video-Recorder #42: How exacly the code records avi?

Yuxekova, 2021-02-27:
I'm trying to implement this for projects with esp-idf, junior.2x does the job but it's a little bit spaghetti.  What exactly is the code doing?

jameszah, 2021-02-28:
You would have to be more specific.

About
Video Recorder for ESP32-CAM with http server for config and ftp server to download video

maritoperezdeargentina, 2021-03-22:
Hi @Yuxekova , I don't know exactly the code works, but can tell you it records a format called MMPEG, 

basically it stores a sequence of JPG files, one after other, the full frames. 

AVI also has a header to identify this format, in the code you can found it, 

but the rest of the process, as far I can understand, should be ask the camera driver for the JPG compressed image, and then append to the avi file. maybe some closing mark is also needed, 

I hope you can learn more about the format mmpeg, and the codes makes more sense for you  (I had no time to investigate further).

closed 2022-02-02 by jameszah