jameszah/ESP32-CAM-Video-Recorder #44: Awakening from sleep

bogdanys, 2021-03-04:
Hi! I would like to implement a method to wake up the ESP 32 Cam module by touching a wire. 
I don't have an infrared sensor, I just have ESP 32 Cam and an Arduino Uno. And I wondering if it is possible to wake up the device by using one of the free pin from the Esp 32 Cam.
Thank you so much for your answer!

jameszah, 2021-03-04:
You want the Uno to wake up the esp32-cam?
Just wire a Uno gpio to the esp32-cam gpio where the pir was supposed to be, and enable the pir code in the esp32-cam software.  Then the Uno cam start and stop the esp32-cam recording.  
The esp32-cam software has capacitive touch code, which is used to disable the pir, but you could re-use it to replace the pir -- turn on the recording with one touch, and turn off the recording with the next touch.  That would take some cutting and pasting in the code.

bogdanys, 2021-03-05:
But at this stage when will it go into hibernation? When will the PIR sensor no longer detect movement? And that wire sensor touch what is used for?

bogdanys, 2021-03-08:
And I also have another issue. I implemented an if statement, if recording == 0 module go into sleep mode for 1 minute. The problem is when it fall asleep the flash turn on and I dont understand why...

closed 2022-02-02 by jameszah