jameszah/ESP32-CAM-Video-Recorder #46: sending to Telegram not working...

chromoxdor, 2021-03-11:
Tried v98 and also v98-wifiman but i get an error when the esp32 tries to send a picture.

Heres the serial output:

`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

---
setup, core 1, priority = 1
                                    
-------------------------------------
ESP-CAM Video Recorder v98x-WiFiMan
 http://espcam.local - to access the camera
-------------------------------------
Boot number: 1
Wakeup was not caused by deep sleep: 0
Get took 2
Good settings in the EPROM 
Internet_Enabled 1
DeepSleepPir 0
record_on_reboot 0
PIRpin 13
PIRenabled 1
framesize 7
repeat_config 100
xspeed 1
gray 0
quality 12
capture_interval 100
total_frames_config 18000
xlength 15
EnableBOT 1
Starting wifi ...
 Disable brownout

BrownOut Regsiter was (in hex) 40028000
*WM: [1] AutoConnect 
*WM: [2] ESP32 event handler enabled 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: sofu
*WM: [3] Using Password: notorrentplease
*WM: [3] WiFi station enable 
*WM: [2] 20000 ms connectTimeout set
*WM: [2] 20000 ms timeout, waiting for connect...
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.1.122
Succesful Connection using WiFiManager
*WM: [3] unloading 
mDNS responder started 'espcam'
Local time: Thu Mar 11 21:04:07 2021

 Enable brownout
Starting sd card ...
SD card mount successfully!
Name: SU01G
Type: SDSC
Speed: 20 MHz
Size: 968MB
SD_MMC Begin: 1
Starting server ...
http task prio: 1
Camera http started
Starting ftp ...
Total space: 967MB
Used space: 790MB
Starting tasks ...
camera, core 1, priority = 2
aviwriter, core 1, priority = 3
ftp, core 0, priority = 4
Starting camera ...
PIRpin = 1, 1, 1, 1, 1,  
Touch T5 = 39, 64, 65, 65, 65,  
Camera Ready! Use 'http://192.168.1.122' to connect
Start a PIR
zcam / avi / ftp / ftp2 / loop 
0 / 0 / 2676 / 2676 / 0  
Internal Total heap 292064, internal Free Heap 90740
SPIRam Total heap   4194140, SPIRam Free Heap   1506140
ChipRevision 1, Cpu Freq 240, SDK Version v3.3.4-432-g7a85334d8
Avi Writer / Camera / Ftp 
2544 / 452 / 3420
----
Total space: 967MB
Used space: 790MB
Nothing deleted, 81.7% disk full
the loop, core 1, priority = 1
Taking a picture for file ...
File saved: /sdcard/20210311/espcam 2021-03-11 21.04.10 svga_Q12_I100_L15_S1.jpg
Taking a picture for telegram...
Sending Photo Telegram, bytes: 10400

Photo telegram failed ><
z 
Total space: 967MB
Used space: 790MB
Nothing deleted, 81.7% disk full
Starting an avi 

File name will be >/sdcard/20210311/espcam 2021-03-11 21.04.11 svga_Q12_I100_L15_S1.avi<

Recording 150 video frames ...`

I testet my token and id with a example provided by the Universal Arduino Telegram Bot library and it worked.

jameszah, 2021-03-11:
https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/ad82bfea33900d36a773640f972644c6f7831e2d/v98/UniversalTelegramBot.cpp#L352

Turn on all the debug serial.print statements, and it will step you through the problem.

Could be firewall, or telegram server gets busy sometimes.

chromoxdor, 2021-03-11:
Thank you for your quick answer.
I did this: 
```
//unmark following line to enable debug mode
#define _debug
```
didn´t work

since i am not the brightest star in the universe i don´t really know what you mean by that:

> https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/ad82bfea33900d36a773640f972644c6f7831e2d/v98/UniversalTelegramBot.cpp#L352
> 
> Turn on all the debug serial.print statements, and it will step you through the problem.
> 
> Could be firewall, or telegram server gets busy sometimes.

jameszah, 2021-03-11:
right, do that, plus recompile, try it again, and look at the serial port for messages ...

...like you see here "connecting to server" ... or messages about things not working ...

Take the "//" off that line getFreeHeap, and see what it prints out 

Something will be printed to lead you to the problem.
```
#ifdef _debug
    Serial.println(F("[BOT Client]Connecting to server"));
#endif
    if (!client->connect(HOST, SSL_PORT)) {
#ifdef _debug
      Serial.println(F("[BOT Client]Conection error"));
#endif
    }
  }
  if (client->connected()) {

    String start_request = "";
    String end_request = "";

    
    //Serial.print("Start: "); Serial.println(ESP.getFreeHeap());
```

chromoxdor, 2021-03-12:
This is what i get regarding telegram:

```
Taking a picture for telegram...
Sending Photo Telegram, bytes: 6947

Send_photo heap before: 90748
[BOT Client]Connecting to server
[BOT Client]Conection error

Send_photo heap after : 90748

Photo telegram failed ><
```

As mentioned before with an example provided by the Universal Arduino Telegram Bot library it works and Host (api.telegram.org) and Port (443) are the same as in your code so i think i can rule out my internet connection as a problem.

Any suggestions?

jameszah, 2021-03-12:
Might be a arduino esp32 library version.  I haven't tried that with current library version 1.05, which is about a week old.  You could try switching back to 1.04 - in Manage Libraries - and select the version.

[image]

Or then it is just the one line of code 

    `if (!client->connect(HOST, SSL_PORT)) {`

So you could switch to Board Wrover module, and turn on Verbose debugging, and you should see a dozen messages from the WiFi system as it is trying to establish the SSL connection.  

[image]

Come to think of it, I think 1.05 added a setInsecure() function to be compatible with 8266.  So try this:

https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/ad82bfea33900d36a773640f972644c6f7831e2d/v98/UniversalTelegramBot.cpp#L41

Existing code:
```
UniversalTelegramBot::UniversalTelegramBot(String token, Client &client) {
  _token = token;
#ifdef ARDUINO_ESP8266_RELEASE_2_5_0
  //client->setInsecure();
#endif
  this->client = &client;
}
```
New code:
```
UniversalTelegramBot::UniversalTelegramBot(String token, Client &client) {
  _token = token;
  client->setInsecure();
  this->client = &client;
}
```
Otherwise you have to get a fancy signature or something to confirm api.telegram.org is not a fake api.telegram.org, but we are not doing banking here!

jameszah, 2021-03-12:
Also, free up some space on that sd card - it says 81% full, but there might be broken files filling up a bunch of the unused space.

chromoxdor, 2021-03-13:
> Might be a arduino esp32 library version. I haven't tried that with current library version 1.05, which is about a week old. You could try switching back to 1.04 - in Manage Libraries - and select the version.
> 
> [image]
> 
This did the trick fpr me! Thank you very much!!!

> Also, free up some space on that sd card - it says 81% full, but there might be broken files filling up a bunch of the unused space.

I did :)

closed 2022-01-13 by jameszah

ichsanputr, 2022-05-24:
I recommend you for use this package [https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/examples/ESP32/SendPhoto/ESP32-Cam/ESP32-Cam.ino](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/blob/master/examples/ESP32/SendPhoto/ESP32-Cam/ESP32-Cam.ino)