jameszah/ESP32-CAM-Video-Recorder #47: AVI file has wrong aspect ratio

SPD13, 2021-03-17:
Hello,
When I try to playback the recorded video on VLC, the aspect ratio is wrong. The video has a vertical format as if it was filmed on an iPhone.
On quicktime, the video is all black. What player have you tested successfully?
Thanks

jameszah, 2021-03-17:
It might be an arduino-esp32 version problem.  It upgraded a couple weeks ago to 1.05, but all that code is for 1.04 including different framesizes, aspect ratios, etc.

You could try the 1.04 version until I scan through it, and look for version problems.  Look here: https://github.com/jameszah/ESP32-CAM-Video-Recorder/issues/46#issuecomment-797799073

I use the VLC - but it plays with most video apps and editors.  I don't seem to have the apple family currently.

Also on the ov2640 camera (the normal one) the ribbon cable comes out of the left of the frame - the top is 90 degrees clockwise.  The ov5640 camera, the ribbon cable comes out of the top of the frame.

danyhm, 2021-06-27:
I have the same problem and it's not because of the version issue.
I'm using arduino-esp32 1.06 and I have only streaming defined (FTP , Bot PIR and touch are disabled)

the stream aspect ratio is fine but the recorded file has wrong aspect ratios. so I did a quick lookup with media info with media codec pack and it seems there's like 2 resolutions involved? 

[image]

jameszah, 2021-06-27:
Yes.  The problem is that I hardcoded in frame sizes in the original program 

original (1.04) ==> cif 5, vga 6, svga 7, uxga 10
new (1.06)       ==> cif 6, vga 8, svga 12, uxga 13

... and I used those numbers on the http command line to re-program the camera.  So I think you are using "10" for uxga (1600x1200), but the 1.06 drivers are giving you xga (1024x768) jpegs ... so you have a uxga avi with xga frames.

The quick solution is to go to manage boards in arduino ide and switch esp32 back to 1.04, and recompile.  It just takes 30 seconds, and then you can switch it back to 1.06 for other projects.

I had one try to switching it to proper symbols for uxga, ..., and I started adding sxga and hd, and then I had to change the command line thing to text "uxga" instead of numbers "10", and then 1.06 needed for more static ram than 1.04, so it would not compile with all the features installed (like telegram, ftp, ...) ... and I gave up for that day.

It all works correctly for 1.06 in https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior which leaves out many of the ram-hungry features and in https://github.com/jameszah/ESP32-CAM-Video-Telegram.  I'll give another go and getting this program squeezed into its memory for the 1.06 version.  

I've been busy switching it over to mp4 rather than avi ... which is not as simple as I thought 😄

jameszah, 2021-06-27:
... also the https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior version has recording and streaming, so that might be all you need.  (no telegram, no ftp, no pir, no deepsleep) and it works with 1.06.

danyhm, 2021-06-27:
Thanks! the junior version should do what I want, however in the meantime I found another project which works as well and has motion detection !
https://github.com/s60sc/ESP32-CAM_MJPEG2SD

your project is great as well!

closed 2022-01-13 by jameszah