jameszah/ESP32-CAM-Video-Recorder #49: ESP32 Wrover Module error?

degagevip, 2021-04-11:
The is Arduino code, with standard setup for ESP32-CAM
    - Board ESP32 Wrover Module
    - Partition Scheme Huge APP (3MB No OTA)

arduino 1.8.13
espressif systems 1.04
v98x-WiFiMan

Error compiling for development board esp32 wrover module

Why not?

jameszah, 2021-04-11:
Maybe you could paste the errors for me to have a look.

degagevip, 2021-04-13:
Thanks to jameszah , 
because I have modified the WiFimanager.h code before! Did not read the printed information carefully.
The bug is fixed. The program is great

ashish-iottive, 2021-06-29:
> Thanks to jameszah ,
> because I have modified the WiFimanager.h code before! Did not read the printed information carefully.
> The bug is fixed. The program is great

Hey @degagevip if you have successfully executed Camera Webserver with WiFi Manager then could you share the Arduino sketch for that only which include the Basic Example Webserver with Wifi Manager.It would be great help.

degagevip, 2021-07-09:
hi,ashish-iottive
In terms of software, I didn't make much improvement. I just added solar charging chip, fill light and 18650 battery to the circuit. I will arrange the PCB and announce it in the near future.

ashish-iottive, 2021-07-09:
Hey @degagevip I have figured out the WiFi Manager library in the ESP32-Camera Webserver sketch, in this sketch I have set an static IP which can be viewed in WiFi Manager portal to remove the use of Serial Monitor and a button to reset the Wi-Fi Manager settings and added mDNS to advertising the IP Address of the camera.

closed 2022-01-13 by jameszah