jameszah/ESP32-CAM-Video-Recorder #51: I can only take photos at a 5 second interval without recording a video ??

Igor7458, 2021-05-01:
I need to take pictures every 5 seconds to make a timelapse ... but I saw that the code makes the recording of videos. Can you help me with this ??? 

i need the tcp functions and the telegram bot still working 

 remembering that i only need the photos every 5 seconds i need to disable video recording so as not to overfill my sd card

jameszah, 2021-05-02:
You want an individual jpg file every 5 seconds, for 12 hours a day, or ~9000 files per day?  That is very laborious to copy them off the sd to the computer, then compose them into a timelapse.

It is much more efficient to stick them together into a mjpeg or avi, and give it a speedup factor (100 times to give you 20 fps playack), so you have 1 file per hour, or 1 file per day, and they will play on a computer without any editing or recomposing.  Or de-compose the the avi into 9000 images to fiddle with them, then recompose them into a timelapse video.

You can set capture_interval = 5000, and it will put them together into an avi, with an index, and your video editor and process it nicely.

Regarding start/end times in the day -- I don't have anything for that, but you have the time, so you could add a line of code at

https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/master/v98/TimeLapseAvi98x.ino
https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/3d4b704455be09ad5a21679d734850f50b9349c0/v98/TimeLapseAvi98x.ino#L1523

when the program wants to start a recording, just check the time, and don't start a recording if the time is night

Igor7458, 2021-05-02:
Ok, I'm sorry for the silly questions but I'm a beginner in the programming world

With your answer I am curious about the setings code

I need one file a day already in timelapse

I have some difficulty understanding some commands like these

-int repeat_config
-volatile int total_frames_config

What do you think?

1 int  framesize = 10;                //  10 UXGA, 7 SVGA, 6 VGA, 5 CIF
2 int  repeat_config = 300;                 //  repaeat same movie this many times
3 int  xspeed = 100;                   //  playback speed - realtime is 1, or 300 means playpack 30 fps of frames at 10 second per frame ( 30 fps / 0.1 fps ) 
4 int  gray = 0;                     //  not gray
5 int  quality = 10;                 //  quality on the 10..50 subscale - 10 is good, 20 is grainy and smaller files, 12 is better in bright sunshine due to clipping
6 int  capture_interval = 5000;       //  milli-seconds between frames
7 volatile int  total_frames_config = 360;  //  how many frames - length of movie in ms is total_frames x capture_interval

jameszah, 2021-05-02:
if you set total_frames_config to 8640 (12 hours x 60 minutes x 12 pictures per minute), then it should produce one 12 hour file  ... you have to put in some code to start it at the correct time.  You could just let it run 24 hours and throw away the night file.

repeat_config is how many of the movies to make.  So just make it large.  At 2 movies per day x 30 days = 60, it will run for a month without intervention and then stop.  You sd will probably fill up before then.  If you have the file delete stuff turned on, you might want to make repeat_config smaller to make sure you are not over writing your old recordings.

uxga quality 10 will produce a jpeg of about 150kb or 200 kb, so 8640 frames will be about 1.3 GB - 1.5 GB ... depending on your light and complexity of the scene.

https://github.com/espressif/esp32-camera/issues/185#issuecomment-705171711

Also reminder that that you have to use esp32-arduino library 1.04, as I have not done the mods for 1.06 the current version).  For example, I used "10" for uxga, but in 1.06 "10"   stands for xga, so some editing required. 😄 

Or switch over to this program which is much simpler -- although has a different menu of features:

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior

Igor7458, 2021-05-02:
Very good, I managed to generate the timelapse

I have some more doubts

-I saw that the generated file is .avi, can I change the .avi for .mp4 ??

-I put Length = 3600 seconds, so every 1 hour a file will be generated, but I wanted the bot to send me pictures every 30 minutes. Can I keep an hour of length and a 30-minute break from the bot photos ??

-The name of the generated files is this:

CAM-01 2021-05-02 11.24.04 uxga_Q10_I5000_L3600_S150

I can remove this part of the generated file name:
 uxga_Q10_I5000_L3600_S150

??

I'm sorry for the amount of silly questions .... your code is sensational !!!!

jameszah, 2021-05-03:
mp4 vs avi

These are the container file formats -- avi is older and simpler than mp4 -- and most everyone can read an avi, and display or edit it, or convert to mp4.  This program only does avi.

length vs bot

The "send to telegram" function uses the highly secure ssl -- secure socket layer -- which uses massive amounts of memory, and I could not run the video recorder and the secure-socket-layer at the sane time, so the solution was to send the first frame, and when that was done (3-5 seconds), then start the recording.  So you could make your video 1 hour long -- 3600 seconds, or 720 frames at 5 sec/frame-- and you will get a bot picture every hour, and 12 files per day.  Which is not too many files to copy and paste them.  I use 2 hour videos with a bot photo for some slow timelapses -- easier to find the time of the part you want.

filenames

Here is the code for file names -- you can delete the different details of framesize quality interval speed etc.

https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/3d4b704455be09ad5a21679d734850f50b9349c0/v98/TimeLapseAvi98x.ino#L1677

```
 time(&now);
  localtime_r(&now, &timeinfo);

  strftime(strftime_buf2, sizeof(strftime_buf2), "/%Y%m%d", &timeinfo);
  SD_MMC.mkdir(strftime_buf2);

  strftime(strftime_buf, sizeof(strftime_buf), "%F %H.%M.%S", &timeinfo);

  if (framesize == 6) {
    sprintf(fname, "/sdcard%s/%s %s vga_Q%d_I%d_L%d_S%d.avi", strftime_buf2, devname, strftime_buf, quality, capture_interval, xlength, xspeed);
  } else if (framesize == 7) {
    sprintf(fname, "/sdcard%s/%s %s svga_Q%d_I%d_L%d_S%d.avi", strftime_buf2, devname,  strftime_buf, quality, capture_interval, xlength, xspeed);
  } else if (framesize == 10) {
    sprintf(fname, "/sdcard%s/%s %s uxga_Q%d_I%d_L%d_S%d.avi", strftime_buf2, devname, strftime_buf, quality, capture_interval, xlength, xspeed);
  } else  if (framesize == 5) {
    sprintf(fname, "/sdcard%s/%s %s cif_Q%d_I%d_L%d_S%d.avi", strftime_buf2, devname, strftime_buf, quality, capture_interval, xlength, xspeed);
  } else {
    Serial.println("Wrong framesize");
  }
```

Igor7458, 2021-05-05:
Well after a long time testing and configuring I came to the conclusion that your code is incredible

everything works except the telegram bot i really don't know what's wrong .... until a few days everything was working without any problem .... the bot sent me pictures every 1 hour without problems

now the bot sometimes sends me and sometimes it fails, i have already reconfigured the settings and i have no idea. Do you have something in mind ??

the bot configuration region looks like this:

RTC_DATA_ATTR int EnableBOT = 0;
#define BOTtoken  "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"   // get your own bot and id at telegram.org 
#define BOTme     "XXXXXXXXXXX"

jameszah, 2021-05-06:
You could try fiddling with these numbers 

```
  int waitForResponse = 5000;  //jz = 1500; 
  int jzdelay = 60;  // delay between multipart blocks
  int jzblocksize = 2 * 1024; // multipart block size
```

When I started using the UniversalTelegramBot.h library, it never worked.  There is a long discussion on that site about adding delays (the 60 above) to prevent a burst of messages at the telegram website -- that are blocked as a potential cyber attack.  I fiddled around with those numbers -- delay and blocksize -- to try to get it to work.  But after a while, it just works without any adjustments to the original Brian Lough code.  Meaning delay = 0 , and blocksize = 512 I think.  So I think it is some subtlety of your local internet speed, and the telegram.org watching you for too many posts, or unusual activity, or just randomness maybe???

You could just add a line of code to see if it fails, and try it again ... and your success rate my jump to close even to 100%.

Igor7458, 2021-05-06:
The messages follow a sending reason. I would say it is an unusual activity

for the correct I would have an image every hour but sometimes the image fails to send and it takes two hours OR MORE to send an image successfully, do you think that a weak connection with the inernet can cause this ??

my internet is not the best:

DOWNLOAD: 300MB / S

UPLOAD: 100MB / S

 but I think this is enough to send a simple image

I will also shave a huge slow to download the videos through the FTP server, it varies from 80 to 100KB / S I believe that this is a bad connection with my router

remembering that I don't have the external antenna ...

I think the best option is to make the new line of code to fail it, try to send it again, the problem is that I don't know how to program heheheh I have to learn how to do it yet

jameszah, 2021-05-08:
That internet is plenty fast.  The normal wifi on esp32 is internet defaults to "half-asleep" mode -- which on my router configures as 6,000 kbps or ~ 0.6 megabytes per seconds.  The code below will turn the modem on fully-awake, which on my router the esp32-router switches up to 72,000 kbps ... or about 10 times faster.

The esp32 is also quite busy, so if you shut off the recording, the ftp will run faster.

[image]

```
#include "esp_wifi.h"
  wifi_ps_type_t the_type;

  esp_err_t get_ps = esp_wifi_get_ps(&the_type);
  Serial.printf("The power save was: %d\n", the_type);

  Serial.printf("Set power save to %d\n", WIFI_PS_NONE);
  esp_err_t set_ps = esp_wifi_set_ps(WIFI_PS_NONE);

  esp_err_t new_ps = esp_wifi_get_ps(&the_type);
  Serial.printf("The power save is : %d\n", the_type);
```

Igor7458, 2021-07-13:
ok after a long time without messing with the project I resumed activities today and everything worked correctly !!! I think about deploying ESP32CAM together with a meteorological monitoring system....do you think it is possible to couple a DHT11 sensor with es32cam and show the temperature information on the homepage of the IP of the esp?? or this would not be possible due to processing or something like that

your code is great for cloud time lapse

jameszah, 2021-07-14:
You could use pin 13 and 12 for other projects, like the dht sensor, which I think is a one wire device.
You just have to strip out the other uses of that pin from the existing code.

I think this program was using 12 for the pir, and 13 for touch sensor.
Also 13 is better than 12, as 12 is checked on boot, and fails if your sensor/circuit is pulling it up.
Also 12 and 13 are still wired to the sd card, so you cannot send too much current into them or it might interfere with the sd.

Otherwise you can use 13 just like it is used here for the pir.

Igor7458, 2021-07-14:
Perfect... one more question

my intention with all this and the timelapse of the cloud movements during the day so the Playback Speed ​​function is essential

the maximum value is 300 correct?

I can put a higher value for the video to be faster ??

jameszah, 2021-07-14:
5 sec / frame = 0.2 frames / sec 

0.2 frames / sec x 300 = 60 fps playback on pc 

60 fps is fine.  24 or 30 fps for normal video, or 60, 120, +++ for video games
 
300 will give you 1 hour realtime in 12 seconds of playback, or 16 hours daylight in 3 minutes

You can change the 300 to anything - it just calculates the micro-seconds between frames, to tell the computer how fast to play the frames.  But the mjpeg has big files, and you could overpower your computer if you make it too fast.  The alternative would be to switch to 10 seconds per frame, etc.

closed 2022-01-13 by jameszah

vipelz, 2022-10-01:
> Regarding start/end times in the day -- I don't have anything for that, but you have the time, so you could add a line of code at [...]

Hi I would like to understand how to start the recording for example at 7 and stop at 19.
Sorry for the noob question but does the code below work?
Also how can I use the configuration file to easily define the start and end time of the recording?

Thank you

 ```
if (now.hour() > 7 && now.hour() < 19 && newfile == 0 && recording == 1) {                         

```