jameszah/ESP32-CAM-Video-Recorder #54: Compile error v98x-WiFiMan - undefined reference to `FtpServer::handleFTP()'

alexw22, 2021-07-05:
Thank you jameszah for this wonderful project! I have been using the v98 version for a couple months now and has been very useful. I like the idea of being able to change the wifi credentials without having to re-flash the esp32. I finally got to updating all my esp32 boards however I am getting the following error on compile. I am happy to provide more info if necessary.

Compile error as it appears in the console:
```
sketch/TimeLapseAvi98x-WiFiMan.ino.cpp.o:(.literal._Z14codeForFtpTaskPv+0x14): undefined reference to `FtpServer::handleFTP()'
sketch/TimeLapseAvi98x-WiFiMan.ino.cpp.o:(.literal._Z5setupv+0xd0): undefined reference to `FtpServer::begin(String, String)'
sketch/TimeLapseAvi98x-WiFiMan.ino.cpp.o: In function `codeForFtpTask(void*)':
Downloads/JamesZah - TimeLapseAvi98x-WiFiMan/TimeLapseAvi98x-WiFiMan/TimeLapseAvi98x-WiFiMan.ino:697: undefined reference to `FtpServer::handleFTP()'
sketch/TimeLapseAvi98x-WiFiMan.ino.cpp.o: In function `setup()':

```
I made sure that I am using all updated libraries and that the esp32 library is version 1.0.4 instead of one of the newer ones.
In trying to overcome this error I tried compiling with older versions of the esp32_ftpclient library. However, the same error would show up.

Compiling v98 without the WiFiMan code results in a successful compile without errors.

jameszah, 2021-07-06:
It must be something to do with folders - make sure all the files are in the same folder, and "readable" -- it seems it cannot find the ftp.cpp file that has that handleFTP code.  You have one folder called "sketch" and another call "Downloads/JamesZah", so put all the files together, or you have to put paths into the include file to tell it where the files are.

[image]

[image]

alexw22, 2021-07-06:
Thank you Jameszah for narrowing down the issue! I managed to solve it. I had all the files in one folder, but after re-copying and pasting the ESP32FtpServer.cpp contents from your GitHub into the arduino IDE ESP32FtpServer.cpp tab, the compile and flash went through!
I must have made an unintentional keystroke copying and pasting the raw file views as Github did not allow me to download only the v98x-WiFiMan subdirectory.

I am really excited about all the updates you have made in the last couple of months.
Thank you for your help and have a wonderful day!
Alex

closed 2021-07-06 by alexw22