jameszah/ESP32-CAM-Video-Recorder #56: problem with recording times

Ak1167, 2021-08-14:
Hi everyone.
I have a esp32-cam AT Thinker module and trying to record video on Micro-SD card by it.
I upload TimeLapseAvi98x.ino version of code and I have problem with recording time !
every recorded video time is about 30 minute.
If module shuts down before this time no data record in video file.
Am I wrong at something ? 

thank's alot.

deltazerorsan, 2021-08-14:
> Hi everyone.
> I have a esp32-cam AT Thinker module and trying to record video on Micro-SD card by it.
> I upload TimeLapseAvi98x.ino version of code and I have problem with recording time !
> every [...]

Hi @Ak1167 that seems the way this soft works. You can record every minute but you need to stop the recording before shutting down the cam or the last fragment will be lost.

Rom

jameszah, 2021-08-14:
You can change the 30 min video length, with compile-time parameters, or with the http interface.

You are correct that it has to close the file to save it properly.  If you unplug it, or lose battery power, the file still exists as a broken file, which you can recover with a diskcheck.  And it will use up disk space until you recover and delete it.

It is a bug, not a feature.  It is corrected the the newer version https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior, which uses the esp disk routines, rather than the posix standard stuff.  On that program the file will be intact and in the file system, but it will not have an index at the end, and may have the playback speed slightly off, as that is updated when the file is closed.  VLC video player will rebuild the index and play it, but you have to modify the playback speed correctly.

I'm updating this version with that improvement and several others - but several projects using my time. 😄

Ak1167, 2021-08-16:
thanks a lot for your helps.
I download junior version and upload it on my module . it works properly .

closed 2022-01-13 by jameszah