jameszah/ESP32-CAM-Video-Recorder #59: Compilation error

mudassar791, 2021-09-23:
ESP32-CAM-Video-Telegram_8.8x.ino:386:19: error: 'class UniversalTelegramBot' has no member named 'sendMultipartFormDataToTelegramWithCaption

Every thing was a mentioned on read me page but still complilatiom failed with above error.

jameszah, 2021-09-23:
https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/master/v99/UniversalTelegramBot.h

Do you have the telegram include .h and .cpp in the same folder as the .ino file?

mudassar791, 2021-09-23:
Yes, both files and .ino file is in same folder.

jameszah, 2021-09-23:
https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/f90207b74c8ac247ef20f3cde9664b5f80844a02/v99/UniversalTelegramBot.h#L72

https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/f90207b74c8ac247ef20f3cde9664b5f80844a02/v99/UniversalTelegramBot.cpp#L352

Must be something to do with your development environment.

dherakles, 2022-01-09:
C:\Users\Dell\Documents\meine Projekte\ESP32-CAM-Video-Recorder-master\v99\TimeLapseAvi99x\TimeLapseAvi99x.ino:140:3: error: conflicting declaration 'typedef enum HTTPMethod HTTPMethod'
 } HTTPMethod;
   ^
In file included from C:\Users\Dell\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WebServer\src/WebServer.h:30:0,
                 from c:\Users\Dell\Documents\Arduino\libraries\WiFiManager/WiFiManager.h:88,
                 from C:\Users\Dell\Documents\meine Projekte\ESP32-CAM-Video-Recorder-master\v99\TimeLapseAvi99x\TimeLapseAvi99x.ino:1:
C:\Users\Dell\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WebServer\src/HTTP_Method.h:6:26: note: previous declaration as 'typedef enum http_method HTTPMethod'
 typedef enum http_method HTTPMethod;
                          ^
Compilation error: exit status 1}
V98, V98 Man without Error compiled - Arduino IDE 2 / ESP32 1.06

dherakles, 2022-01-09:
sorry, falsches Thema

closed 2022-01-13 by jameszah