jameszah/ESP32-CAM-Video-Recorder #61: SDCARD

farisanaklumajang, 2022-01-24:
- Jika sdcard full, video lama  tidak otomatis hapus,
   Esp32 juga  malah macet

jameszah, 2022-01-24:
Do you have this update:

[image]

https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/master/vA1/TimeLapseAviA1.5x.ino

https://github.com/jameszah/ESP32-CAM-Video-Recorder/commit/58dc7e9e64399e3efdad4af8c9db3a06b8ab7493

farisanaklumajang, 2022-01-24:
Saya sudah memakai TImelapseAviA1.5x.ino,,,

farisanaklumajang, 2022-01-24:
[image]

RESTART SENDIRI ESP32NYA, TANPA MEREKAM ( KARNA TIDAK BISA MENGHAPUS FILE REKAMAN OLD )

jameszah, 2022-01-24:
I made the changes for that file delete problem about a week ago.
It was with a new version of the file 1.5x.

I have now changed the naming convention and called it 1x

This is the file you need:

https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/master/vA1/TimeLapseAviA1x.ino

Old versions can be seen here:

https://github.com/jameszah/ESP32-CAM-Video-Recorder/commits/master/vA1

farisanaklumajang, 2022-01-24:
Oke siap,
Nanti akan saya coba 👍

farisanaklumajang, 2022-01-25:
[image]

Remove dir filed

jameszah, 2022-01-25:
Try a quick-format in fat32 on a windows or mac.

You might have broken files that are filling up space that appears available.

And the root is "/sdcard" so my computer shows 

/sdcard/20220125/snolens_2022-01-25_00.01.51_vga_Q12_I200_L1800_S1.avi

... so that unprintable ascii chars might mess it up.

https://github.com/espressif/arduino-esp32/blob/6a7bcabd6b7a33f074f93ed60e5cc4378d350b81/libraries/SD_MMC/src/SD_MMC.h#L41

closed 2022-02-04 by farisanaklumajang