jameszah/ESP32-CAM-Video-Recorder #62: Camera Failure

TSKR007, 2022-01-26:
i am new to this.....
so there is an issue on serial Monitor its showing this.....

_Internet connected
mDNS responder started 'desklens'
Local time: Wed Jan 26 14:32:23 2022

The power save was: 1
Set power save to 0
The power save is : 0
 Enable brownout
Starting sd card ...
SD card mount successfully!
Name: USD  
Type: SDHC/SDXC
Speed: 40 MHz
Size: 30720MB
SD_MMC Begin: E (4661) vfs_fat_sdmmc: host init failed (0x103).
1
Starting server ...
http task prio: 5
Camera http started
Starting ftp ...
Starting the esp32 sd file manager
Total space: 30700MB
Used space: 0MB
Starting tasks ...
camera, core 1, priority = 2
aviwriter, core 1, priority = 3
ftp, core 1, priority = 4
Starting camera ...
E (4868) cam_hal: cam_dma_config(280): frame buffer malloc failed
E (4869) cam_hal: cam_config(364): cam_dma_config failed
E (4869) camera: Camera config failed with error 0xffffffff
Camera init failed with error 0xffffffff_

jameszah, 2022-01-26:
E (4868) cam_hal: cam_dma_config(280): frame buffer malloc failed

What kind of board do you have?  

You need an ai-thinker esp32-cam board or another with 4 MB of psram.

There is not enough memory psram to allocate the camera buffers for jpeg processing.

You could add this print_ram() line here

https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/1d387ab82d1488b789be37dd6886dbc7ab963057/vA1/TimeLapseAviA1x.ino#L1255

cameraWebServer avoids the problem by making a small frame, and slowing down the camera.

https://github.com/espressif/arduino-esp32/blob/6a7bcabd6b7a33f074f93ed60e5cc4378d350b81/libraries/ESP32/examples/Camera/CameraWebServer/CameraWebServer.ino#L57

TSKR007, 2022-01-26:
Hi sir,
thanks for the reply..i am using ai thinker cam ESP32-s
[image]
after adding that line i got this....

_Internet connected
mDNS responder started 'desklens'
Local time: Wed Jan 26 18:03:00 2022

The power save was: 1
Set power save to 0
The power save is : 0
 Enable brownout
Starting sd card ...
SD card mount successfully!
Name: USD  
Type: SDHC/SDXC
Speed: 40 MHz
Size: 30720MB
SD_MMC Begin: E (4663) vfs_fat_sdmmc: host init failed (0x103).
1
Starting server ...
http task prio: 5
Camera http started
Starting ftp ...
Starting the esp32 sd file manager
Total space: 30700MB
Used space: 0MB
Starting tasks ...
camera, core 1, priority = 2
aviwriter, core 1, priority = 3
ftp, core 1, priority = 4
cam / avi / ftp / ftp2 / loop 
0 / 0 / 1 / 0 / 0  
Internal Total heap 290076, internal Free Heap 208552
SPIRam Total heap   2094495, SPIRam Free Heap   2048223
ChipRevision 1, Cpu Freq 240, SDK Version v4.4-beta1-189-ga79dc75f0a
----
Starting camera ...
E (6879) cam_hal: cam_dma_config(280): frame buffer malloc failed
E (6880) cam_hal: cam_config(364): cam_dma_config failed
E (6880) camera: Camera config failed with error 0xffffffff
Camera init failed with error 0xffffffff_

jameszah, 2022-01-26:
Looks like you got a esp32-cam with 2 MB of psram.
That circle-lightning-Y brand.

[image]
[image]

I've always observed 4MB on my non-AI modules, but I see AI-Thinker says they have 8 MB,

[image]

[image]

You might be able to modify the code to use less psram ... but it would be some work, and would only operate with a small frame size, and slow recording.

closed 2022-01-29 by jameszah