jameszah/ESP32-CAM-Video-Recorder #63: major fail

MRHERT, 2022-01-29:
when i upload the code and finish then i open serial monitor it gives le major fail 1/10
majort fail 2/10 ...... how i can solve this please

jameszah, 2022-01-29:
What does it say before major fail?
You can cut and paste it in here.

 It is either the sd card missing or bad, gpios grounded, camera missing, loose, or broken, wifi missing or broken, ...

MRHERT, 2022-01-29:
> What does it say before major fail?
> You can cut and paste it in here.
>
> It is either the sd card missing or bad, gpios grounded, camera missing,
> loose, or broken, wifi missing or broken, ...
>
> or Android

Can i conract you from your whatssapp please so that i can send you
pictures and explain more please its urgent

MRHERT, 2022-01-29:
or just contact me in my email please yassine.blog1@gmail.com

jameszah, 2022-01-29:
You have no psram, or it is not working.

You need a esp32-cam board with 4mb of psram. 

E (162) psram: PSRAM ID read error: 0xffffffff

https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/b6c9af8970bc2321935662286603da7ce6484271/vA1/TimeLapseAviA1x.ino#L1166

MRHERT, 2022-01-29:
so what is the solution i tried to do this with differents projects and it did work but for this recording project no is there any solution ?

MRHERT, 2022-01-29:
it has a 4Mb of psram im sure but what is the issue

closed 2022-01-30 by jameszah

MRHERT, 2022-01-30:
NO SOLUTION ? OKAY THANK YOU

MnemonicFish, 2023-01-17:
you need to enable PSRAM from the Arduino IDE menu, Tools > PSRAM > Enabled

[image]