jameszah/ESP32-CAM-Video-Recorder #67: Waveshare ESP32 One

derbmann, 2022-04-20:
Hi,

I got the [Waveshare ESP32 One ](https://www.waveshare.com/wiki/ESP32_One) and i wanted to use your program on it.
Sadly i got a bit of trouble with it, most likely because I'm a beginner...

I did change the camera pins to.: 
> #define PWDN_GPIO_NUM    -1
#define RESET_GPIO_NUM   -1
#define XCLK_GPIO_NUM    4
#define SIOD_GPIO_NUM    18
#define SIOC_GPIO_NUM    23
#define Y9_GPIO_NUM      36
#define Y8_GPIO_NUM      37
#define Y7_GPIO_NUM      38
#define Y6_GPIO_NUM      39
#define Y5_GPIO_NUM      35
#define Y4_GPIO_NUM      14
#define Y3_GPIO_NUM      13
#define Y2_GPIO_NUM      34
#define VSYNC_GPIO_NUM   5
#define HREF_GPIO_NUM    27
#define PCLK_GPIO_NUM    25

(Because the CameraWebServer example works with these settings.)

But the actual problem is now the SD card.
The card reader is connected to.:

> SD Pin -> ESP32 Pin
1 -> NC
2 -> IO15
3 -> IO13
4 -> VDD_SDIO (3,3V)
5 -> IO14
6 -> GND
7 -> IO12
8 -> NC
9 -> GND

This means its connected to SPI in the HSDPI pinout.
Since i got the SD_Test sketch working ill think I'm right to this point.

But since you are using the sd_mmc i can't use the card reader with it, right?

Just for testing purpose I desoldered the card reader and connected the following pins to it.:
> SD Pin -> ESP32 Pin
1 -> NC
2 -> NC
3 -> IO15
4 -> VDD_SDIO (3,3V)
5 -> IO14
6 -> GND
7 -> IO2
8 -> NC
9 -> GND

The SDMMC_Test sketch is working with these connected pins in the 1-bit mode SD_MMC.begin("/cdcard", true);.
And after adding pinMode(2, PULLUP); to line ~1457 at static esp_err_t init_sdcard() your program finally was able to read the sd card but fails later again with some sd error on the startup.
Startup-Log & Settings:
[startup.txt](https://github.com/jameszah/ESP32-CAM-Video-Recorder/files/8525038/startup.txt)
[TimeLapseAviA1x.ino.txt](https://github.com/jameszah/ESP32-CAM-Video-Recorder/files/8525101/TimeLapseAviA1x.ino.txt)
[settings.h.txt](https://github.com/jameszah/ESP32-CAM-Video-Recorder/files/8525106/settings.h.txt)

I'm now at the end of my knowledge and need your/someones help pls.
What did I wrong or need to change to get it working?

Thank you,
derbmann

jameszah, 2022-04-21:
Are you re-using pin 14 - camera and sd?

I'll have to compare the schematics to see which data pins the sd uses in 1-bit mode.
The unused pins could be left hanging to prevent interaction. I think gpio4, 12, 13 are the unsed pins on the esp32-cam ai-thinker module sd card connection and gpio2 is the 1-bit.

So I think you are also using gpio2 for the data0, and your data1 and data2 are disconnected from the sd connector (the red X).  And cd/data3 was gpio15 and is now disconnected.

You could disconnect 14 and 15 and try it.  The camera might be turning on 14, which is disturbing the sd card.

[image]

[image]

jameszah, 2022-04-21:
gpio14 is h2s_clk, spi_clk,  MTMS?  

You probably need that one, so you would have to tell sd_mmc to use another gpio for that one, as I don't imagine you could move a gpio from the camera connector to use another pin for the gpio14 on the camera.

[image]
[image]

jameszah, 2022-04-30:
When using the camera gpio14 is an input to the esp32 (y4 - data line), when using the sdcard gpio14 is an output (sclk clock)
so independently they will work, but not at the same time.

[image]