jameszah/ESP32-CAM-Video-Recorder #70: ESP32-CAM-Video-Recorder-junior v60.4.5 Reboot Problem

shahbai, 2022-09-02:
**hi
When I install version 60.4.5, I get reboots again and again:**

`-------------------------------------
ESP32-CAM-Video-Recorder-junior v60.4.5
-------------------------------------
setup, core 1, priority = 1
--- reboot ------ because: ESP_RST_PANIC
Good settings in the EPROM 
New File Group 89
Writing to EPROM ...
Mounting the SD card ...
SD_MMC Begin: 1
SD_MMC Card Type: SDSC
SD_MMC Card Size: 1886MB
Try to get parameters from config.txt ...
Reading config.txt
=========   Data fram config.txt and defaults  =========
Name desklens
Framesize 11
Quality 12
Framesize config 13
Quality config 5
Buffers config 4
Length 1800
Interval 0
Speedup 1
Streamdelay 0
Internet 4
Zone len 3, GMT
ssid A.S.H
pass adhbjrsdg
 desklens_D4
 gb800FD4jz60
Creating logfile /desklens89.999.txt
Setting up the camera ...
Frame config 13, quality config 5, buffers config 4
Before camera config ...Internal Total heap 290288, internal Free Heap 263568, SPIRam Total heap   2094963, SPIRam Free Heap   2068027
After  camera config ...Internal Total heap 290048, internal Free Heap 226532, SPIRam Total heap   2094915, SPIRam Free Heap   531979

Camera started correctly, Type is 26 (hex) of 9650, 7725, 2640, 3660, 5640

Pic, len=29663, new fb 3F8071CC
Pic, len=29522, new fb 3F864DDC
Pic, len=75583, new fb 3F8C29EC
Pic, len=74410, new fb 3F9205FC
Pic, len=73284, new fb 3F8071CC
Pic, len=56573, new fb 3F864DDC
Pic, len=57607, new fb 3F8071CC
Pic, len=57579, new fb 3F864DDC
Pic, len=57886, new fb 3F8071CC
Pic, len=57819, new fb 3F864DDC
End of setup ...Internal Total heap 290048, internal Free Heap 226532, SPIRam Total heap   2094915, SPIRam Free Heap   531979
Checking SD for available space ...
Total space: 1877MB
Used space: 0MB
Nothing deleted, 0.0% disk full
Creating the_camera_loop_task
the camera loop, core 0, priority = 6
the_sd_loop, core 1, priority = 4

low prio streaming loop, core 1, priority = 3
Starting the streaming
  End of setup()

Starting the wifi ...
ssid >A.S.H<, pass >adhbjrsdg<

Start the avi ... at 3034
Framesize 11, quality 12, length 1800 seconds

Starting an avi 
File open: /desklens89.001.avi

Recording 1800 seconds.
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008bbc0  PS      : 0x00060230  A0      : 0x800d6a8c  A1      : 0x3ffb4540  
A2      : 0x00000000  A3      : 0x3f864ddc  A4      : 0x0000e3a8  A5      : 0x00000000  
A6      : 0xe0ffd8ff  A7      : 0x464a1000  A8      : 0x00000000  A9      : 0x3ffb44f0  
A10     : 0x3ffc8b58  A11     : 0x3ffbf188  A12     : 0x3ffc8b58  A13     : 0x00060223  
A14     : 0x00060220  A15     : 0x00000001  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4008bbbc  LEND    : 0x4008bbd8  LCOUNT  : 0x00000e39  

Backtrace:0x4008bbbd:0x3ffb45400x400d6a89:0x3ffb4550 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0

---
                                    
-------------------------------------
ESP32-CAM-Video-Recorder-junior v60.4.5
-------------------------------------
setup, core 1, priority = 1
--- reboot ------ because: ESP_RST_PANIC
Good settings in the EPROM 
New File Group 90
Writing to EPROM ...
Mounting the SD card ...
SD_MMC Begin: 1
SD_MMC Card Type: SDSC
SD_MMC Card Size: 1886MB
Try to get parameters from config.txt ...
Reading config.txt
=========   Data fram config.txt and defaults  =========
Name desklens
Framesize 11
Quality 12
Framesize config 13
Quality config 5
Buffers config 4
Length 1800
Interval 0
Speedup 1
Streamdelay 0
Internet 4
Zone len 3, GMT
ssid A.S.H
pass adhbjrsdg
 desklens_D4
 gb800FD4jz60
Creating logfile /desklens90.999.txt
Setting up the camera ...
Frame config 13, quality config 5, buffers config 4
Before camera config ...Internal Total heap 290288, internal Free Heap 263568, SPIRam Total heap   2094963, SPIRam Free Heap   2068027
After  camera config ...Internal Total heap 290048, internal Free Heap 226532, SPIRam Total heap   2094915, SPIRam Free Heap   531979

Camera started correctly, Type is 26 (hex) of 9650, 7725, 2640, 3660, 5640

Pic, len=136240, new fb 3F8071CC
Pic, len=29077, new fb 3F864DDC
Pic, len=28773, new fb 3F8C29EC
Pic, len=74404, new fb 3F9205FC
Pic, len=72243, new fb 3F8071CC
Pic, len=59785, new fb 3F864DDC
Pic, len=60726, new fb 3F8071CC
Pic, len=61095, new fb 3F864DDC
Pic, len=60801, new fb 3F8071CC
Pic, len=60866, new fb 3F864DDC
End of setup ...Internal Total heap 290048, internal Free Heap 226532, SPIRam Total heap   2094915, SPIRam Free Heap   531979
Checking SD for available space ...
Total space: 1877MB
Used space: 0MB
Nothing deleted, 0.0% disk full
Creating the_camera_loop_task
the camera loop, core 0, priority = 6
the_sd_loop, core 1, priority = 4

low prio streaming loop, core 1, priority = 3
Starting the streaming
  End of setup()

Starting the wifi ...
ssid >A.S.H<, pass >adhbjrsdg<

Start the avi ... at 3040
Framesize 11, quality 12, length 1800 seconds

Starting an avi 
File open: /desklens90.001.avi

Recording 1800 seconds.
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008bbc0  PS      : 0x00060a30  A0      : 0x800d6a8c  A1      : 0x3ffb4540  
A2      : 0x00000000  A3      : 0x3f864ddc  A4      : 0x0000ed85  A5      : 0x00000000  
A6      : 0xe0ffd8ff  A7      : 0x464a1000  A8      : 0x00000000  A9      : 0x3ffb44f0  
A10     : 0x3ffc8b58  A11     : 0x3ffbf188  A12     : 0x3ffc8b58  A13     : 0x00060a23  
A14     : 0x00060a20  A15     : 0x00000001  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4008bbbc  LEND    : 0x4008bbd8  LCOUNT  : 0x00000ed7  

Backtrace:0x4008bbbd:0x3ffb45400x400d6a89:0x3ffb4550 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0

---
                                    
-------------------------------------
ESP32-CAM-Video-Recorder-junior v60.4.5
-------------------------------------
setup, core 1, priority = 1
--- reboot ------ because: ESP_RST_PANIC
Good settings in the EPROM 
New File Group 91
Writing to EPROM ...
Mounting the SD card ...
SD_MMC Begin: 1
SD_MMC Card Type: SDSC
SD_MMC Card Size: 1886MB
Try to get parameters from config.txt ...
Reading config.txt
=========   Data fram config.txt and defaults  =========
Name desklens
Framesize 11
Quality 12
Framesize config 13
Quality config 5
Buffers config 4
Length 1800
Interval 0
Speedup 1
Streamdelay 0
Internet 4
Zone len 3, GMT
ssid A.S.H
pass adhbjrsdg
 desklens_D4
 gb800FD4jz60
Creating logfile /desklens91.999.txt
Setting up the camera ...
Frame config 13, quality config 5, buffers config 4
Before camera config ...Internal Total heap 290288, internal Free Heap 263568, SPIRam Total heap   2094963, SPIRam Free Heap   2068027
After  camera config ...Internal Total heap 290048, internal Free Heap 226532, SPIRam Total heap   2094915, SPIRam Free Heap   531979

Camera started correctly, Type is 26 (hex) of 9650, 7725, 2640, 3660, 5640

Pic, len=153547, new fb 3F8071CC
Pic, len=28688, new fb 3F864DDC
Pic, len=28417, new fb 3F8C29EC
Pic, len=72416, new fb 3F9205FC
Pic, len=71813, new fb 3F8071CC
Pic, len=61996, new fb 3F864DDC
Pic, len=62802, new fb 3F8071CC
Pic, len=62000, new fb 3F864DDC
Pic, len=61736, new fb 3F8071CC
`

**Please help me to solve this problem.**

jameszah, 2022-09-02:
[image]

You only have 2 MB of spiram and you need 4MB.  
I suppose I could print out a clear error message in that case.

One of these, or a generic.
 
[image]

shahbai, 2022-09-02:
It means that the code does not run on my device? there's no way?

SanderVerdiesen2, 2022-12-02:
I get the exact same error, but I do have enough SPIRAM. Any idea what the problem could be?

Any help is much appreciated!

---------------------------------------------------------------------------------------------------------------
---
                                    
-------------------------------------
ESP32-CAM-Video-Recorder-junior v60.4.7
-------------------------------------
setup, core 1, priority = 1
--- reboot ------ because: ESP_RST_PANIC
Good settings in the EPROM 
New File Group 124
Writing to EPROM ...
Mounting the SD card ...
SD_MMC Begin: 1
SD_MMC Card Type: SDHC
SD_MMC Card Size: 3781MB
Try to get parameters from config.txt ...
Reading config.txt
=========   Data fram config.txt and defaults  =========
Name desklens
Framesize 11
Quality 12
Framesize config 13
Quality config 5
Buffers config 4
Length 1800
Interval 0
Speedup 1
Streamdelay 0
Internet 5
Zone len 3, GMT
ssid ap
pass mrpeanut
>>>>>>>>>>>>>>>>>>>>> desklens_84
>>>>>>>>>>>>>>>>>>>>> g85c2684jz60
Creating logfile /desklens124.999.txt
Setting up the camera ...
Frame config 13, quality config 5, buffers config 4
Before camera config ...Internal Total heap 303148, internal Free Heap 248164, SPIRam Total heap   4194252, SPIRam Free Heap   4194252
After  camera config ...Internal Total heap 302540, internal Free Heap 190116, SPIRam Total heap   4194188, SPIRam Free Heap   354188

Camera started correctly, Type is 26 (hex) of 9650, 7725, 2640, 3660, 5640

Pic, len=83883, new fb 3FABF254
Pic, len=87868, new fb 3F9D4C44
Pic, len=86513, new fb 3F8EA634
Pic, len=87612, new fb 3F800024
Pic, len=86641, new fb 3FABF254
Pic, len=87066, new fb 3F9D4C44
Pic, len=90355, new fb 3F8EA634
Pic, len=90459, new fb 3F800024
Pic, len=88397, new fb 3FABF254
Pic, len=90920, new fb 3F9D4C44
End of setup ...Internal Total heap 302540, internal Free Heap 190116, SPIRam Total heap   4194188, SPIRam Free Heap   354188
Checking SD for available space ...
Total space: 3777MB
Used space: 5MB
Nothing deleted, 0.1% disk full
Creating the_camera_loop_task
the camera loop, core 0, priority = 6
the_sd_loop, core 1, priority = 4

low prio streaming loop, core 1, priority = 3
Starting the streaming
  End of setup()

Starting the wifi ...
Setting AP (Access Point)…AP IP address: 192.168.4.1
IP: 192.168.4.1
 
mDNS responder started 'desklens'
The power save is : 0
Starting Web Services ...
http task prio: 5
Camera http started
http Stream task prio: 5
Stream 81 http started
http Stream task prio: 5
Stream 82 http started

Disconnect Pin 12 from GND to start recording.

Start the avi ... at 5864
Framesize 11, quality 12, length 1800 seconds

Starting an avi 
File open: /desklens124.001.avi

Recording 1800 seconds.
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4008e1b8  PS      : 0x00060f30  A0      : 0x800d53e0  A1      : 0x3ffdbb20  
A2      : 0x00000000  A3      : 0x3f8ea634  A4      : 0x0001c34b  A5      : 0x00000000  
A6      : 0xe0ffd8ff  A7      : 0x464a1000  A8      : 0x00000000  A9      : 0x3ffdbad0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060f20  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4008e1b4  LEND    : 0x4008e1d0  LCOUNT  : 0x00001c33  

ELF file SHA256: 0000000000000000

Backtrace: 0x4008e1b8:0x3ffdbb20 0x400d53dd:0x3ffdbb30 0x40092096:0x3ffdbba0

Rebooting...

---