jameszah/ESP32-CAM-Video-Recorder #71: Photo telegram failed

DJprasenjit, 2022-09-24:
File saved: /sdcard/20220924/desklens 2022-09-24 23.31.27 uxga_Q12_I10000_L3600_S300.jpg
Taking a picture for telegram...
Sending Photo Telegram, bytes: 53725

Photo telegram failed ><
z 
Starting an avi 

any version telegram issue 
please help me

DJprasenjit, 2022-09-24:
please help me

DJprasenjit, 2022-09-26:
Sending Photo Telegram, bytes: 5310
zzE (49181) MDNS_Networking: Cannot allocate memory (line: 123, free heap: 1506240 bytes)
E (49182) MDNS_Networking: Cannot allocate memory (line: 123, free heap: 1506240 bytes)
E (49396) wifi:rxa_eb mem fail tid=1
zzE (51396) wifi:rxa_eb mem fail tid=1
zzE (53396) wifi:rxa_eb mem fail tid=1
zE (54198) MDNS_Networking: Cannot allocate memory (line: 123, free heap: 1506240 bytes)
E (54200) MDNS_Networking: Cannot allocate memory (line: 123, free heap: 1506240 bytes)
zzzzzE (59114) MDNS_Networking: Cannot allocate memory (line: 123, free heap: 1506240 bytes)
E (59115) MDNS_Networking: Cannot allocate memory (line: 123, free heap: 1506240 bytes)
zzE (61571) MDNS_Networking: Cannot allocate memory (line: 123, free heap: 1506240 bytes)
z#zzzzz#zzz
Send_photo heap after : 74012

Photo telegram failed ><
z 

i tried many method but same fail. i send you the detail of errors

jameszah, 2022-09-28:
What esp32-arduino are you using?  This was done for esp32-arduino 2.02

You seems to be running out of regular ram.
You could try shutting off the detailed debug messages which swallow up ram.
Or try smaller frame sizes, lower quality photos to free up ram.

DJprasenjit, 2022-10-01:
thanks for reply sir
i am using latest version of esp32 board 2.0.6
arduinojson latest version
i compiled all new and old version from v94 to VA1 just edited the settings.h wifi cred. and Bot Cred. and Timezone and uploaded and checked all versions. video recording pir motion sd card all working fine.
but telegram bot isnt working that error comes .
but once the 1st time i upload the vA1 version only once telegram bot sent the image.
after that i thing i messed somthing up in library or somthing else. 
is the error from wifimanager and mDNS ?
i am new in this , know few coding, please help me fix it,
i tried to modify this code v94 and succeded with sending a message to bot in void setup. but in loop nothing works.

DJprasenjit, 2022-10-01:
i lower the picture quality 
but same issue

Taking a picture for telegram...
Sending Photo Telegram, bytes: 4019

Send_photo heap before: 88964

Send_photo heap after : 88964

Photo telegram failed ><
z 
Total space: 30424MB

jameszah, 2022-10-01:
You may have your telegram name or password wrong, or telegram might be blocked, as happens sometimes, by governments or internet service providers.

https://github.com/jameszah/ESP32-CAM-Video-Recorder/blob/b6c9af8970bc2321935662286603da7ce6484271/vA1/UniversalTelegramBot.h#L40

Uncommment the _debug, and see what the detailed messages show.

DJprasenjit, 2022-10-02:
Sending Photo Telegram, bytes: 5256
[BOT Client]Connecting to server
[BOT Client]Conection error

Photo telegram failed ><

connection error coming.
sir i uploaded another telegram bot code on it. that works fine with same Bot credential . 
but this one is not working

DJprasenjit, 2022-10-02:
/*******************************************************************
This code is working fine. telegram bot responding without error, i tried to compare and tried combine both but fail. because i liked these functions like ask esp32cam to send a picture and status and enable disable PIR
but this doesnot have the sd card recording function like your code. so i wanted both functions in one code. can you please help me do it. it would be very helpful. here is the code that works everything...
 ESP32-CAM-Video-Telegram

  This program records an mjpeg avi video in the psram of a ESP32-CAM, and sends a jpeg and a avi video to Telegram.

  https://github.com/jameszah/ESP32-CAM-Video-Telegram is licensed under the
    GNU General Public License v3.0

  by James Zahary  June 1, 2020
  jamzah.plc@gmail.com

  The is Arduino code, with standard setup for ESP32-CAM
    - Board ESP32 Wrover Module
    - Partition Scheme Huge APP (3MB No OTA)
    - or with AI Thinker ESP32-CAM
  Compiled with Arduino 1.8.13 and arduino-esp32 1.0.6 (latest release version) on Jun 10, 2021

  Jan 18, 2022 ver 8.9  
  - updates from Arduino 1.8.19 
    - return from void problem re-runs the function if you dont do a return ???
      https://stackoverflow.com/questions/22742581/warning-control-reaches-end-of-non-void-function-wreturn-type
  - updates for esp32-arduino 2.0.2
    - bug with 2.0.2 handshake timeout - added timeout resets in this file as a workaround
      https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/issues/270#issuecomment-1003795884
   - updates for esp32-arduino 2.0.2
     - esp-camera seems to have changed to fill all free fb buffers in sequence, so must empty them to get a snapshot
     
  Based on these two:

  https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior
  https://github.com/jameszah/ESP32-CAM-Video-Recorder

  and using a modified old version of:
  https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot 

~~~~~~~~~~~~

Using library WiFi at version 2.0.0 in folder: C:\ArduinoPortable\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.2\libraries\WiFi 
Using library WiFiClientSecure at version 2.0.0 in folder: C:\ArduinoPortable\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.2\libraries\WiFiClientSecure 
Using library ArduinoJson at version 6.18.5 in folder: C:\ArduinoPortable\arduino-1.8.19\portable\sketchbook\libraries\ArduinoJson 
Using library ESPmDNS at version 2.0.0 in folder: C:\ArduinoPortable\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.2\libraries\ESPmDNS 
"C:\\ArduinoPortable\\arduino-1.8.19\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\gcc8_4_0-esp-2021r2/bin/xtensa-esp32-elf-size" -A "C:\\Users\\James\\AppData\\Local\\Temp\\arduino_build_57156/ESP32-CAM-Video-Telegram_8.9.ino.elf"
Sketch uses 956193 bytes (30%) of program storage space. Maximum is 3145728 bytes.
Global variables use 63080 bytes (19%) of dynamic memory, leaving 264600 bytes for local variables. Maximum is 327680 bytes.
C:\ArduinoPortable\arduino-1.8.19\portable\packages\esp32\tools\esptool_py\3.1.0/esptool.exe --chip esp32 --port COM7 --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0xe000 C:\ArduinoPortable\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.2/tools/partitions/boot_app0.bin 0x1000 C:\Users\James\AppData\Local\Temp\arduino_build_57156/ESP32-CAM-Video-Telegram_8.9.ino.bootloader.bin 0x10000 C:\Users\James\AppData\Local\Temp\arduino_build_57156/ESP32-CAM-Video-Telegram_8.9.ino.bin 0x8000 C:\Users\James\AppData\Local\Temp\arduino_build_57156/ESP32-CAM-Video-Telegram_8.9.ino.partitions.bin 

 
*******************************************************************/

/*******************************************************************
  -  original opening from Brian Lough telegram bot demo
  
   A Telegram bot for taking a photo with an ESP32Cam

   Parts used:
   ESP32-CAM module* - http://s.click.aliexpress.com/e/bnXR1eYs

    = Affiliate Links

   Note:
   - Make sure that you have either selected ESP32 Wrover Module,
           or another board which has PSRAM enabled
   - Choose "Huge App" partion scheme

   Some of the camera code comes from Rui Santos:
   https://randomnerdtutorials.com/esp32-cam-take-photo-save-microsd-card/

   Written by Brian Lough
[... 1084 lines cut ...]
  esp_camera_fb_return(vid_fb);
  bot.longPoll =  0;
  digitalWrite(33, HIGH);          // light oFF
  if (!avi_enabled) active_interupt = false;
}

void send_the_video() {
  digitalWrite(33, LOW);          // light on
  Serial.println("\n\n\nSending clip with caption");
  Serial.println("\n>>>>> Sending as 512 byte blocks, with a caption, and with jzdelay of 0, bytes=  " + String(psram_avi_ptr - psram_avi_buf));
  avi_buf = psram_avi_buf;

  avi_ptr = 0;
  avi_len = psram_avi_ptr - psram_avi_buf;

  String sent2 = bot.sendMultipartFormDataToTelegramWithCaption("sendDocument", "document", strftime_buf,
                 "image/jpeg", "Intruder alert!", chat_id, psram_avi_ptr - psram_avi_buf,
                 avi_more, avi_next, nullptr, nullptr);

  Serial.println("done!");
  digitalWrite(33, HIGH);          // light off

  bot.longPoll = 5;
  active_interupt = false;
}

jameszah, 2022-10-02:
You seem to be low on memory - when you send to telegram you have 88kb free - below I have 156kb free.  SSL needs lots of memory to talk to telegram - you have used an extra 70kb somewhere.

Do you have an AI-Thinker with ESP32S board, and this arduino setup?

[image]

Might be "core debug level: none" is the problem - all the code to print out the debug error messages

```
Taking a picture for file ...
zFile saved: /sdcard/20221002/desklens_2022-10-02_14.04.18_vga_Q12_I100_L1800_S1.jpg
Taking a picture for telegram...
Sending Photo Telegram, bytes: 16956

Send_photo heap before: 156020
z****z**z*******z
Send_photo heap after : 153936

Photo telegram was successfully sent 
z 
Total space: 30238MB
Used space: 772MB
Nothing deleted, 2.6% disk full
Starting an avi 
```

DJprasenjit, 2022-10-03:
i have the ai tinker esp32cam board but i am selecting the esp32 woover module in the board menu. do i need to select ai tinker esp32 cam?

and what i need to select in core debug level? its normally none.

jameszah, 2022-10-03:
Ai-thinker board sets the parameters correctly.  Wrover board gives you
lots of options to set things wrong and use up your memory.  Set the debug
to none.

DJprasenjit, 2022-10-06:
sendPostMessage: SEND Post Message: {"chat_id":"5108510368","text":"\n  ESP32-CAM Video Recorder v98\n desklens 192.168.0.104  \n Used 356 MB / 30424 MB, Rssi -42\n\n \n  "}
[BOT Client]Connecting to server
[BOT Client]Conection error

[BOT Client]Connecting to server
[BOT Client]Conection error

see bro i am sending a normal text and its showing error

jameszah, 2022-10-06:
I have Arduino 1.8.19 with esp32-arduino 2.0.4.
You have to resolve the extra 70kb for ram you are using before ssl and telegram will work.

You could add some debug statements that print this "Send_photo heap before: 156020"

Are you using AI-Thinker ESP32-CAM with ESP32S written on the steel can?

DJprasenjit, 2022-10-07:
> I have Arduino 1.8.19 with esp32-arduino 2.0.4. You have to resolve the extra 70kb for ram you are using before ssl and telegram will work.
> 
> You could add some debug statements that print this "Send_photo heap before: 156020"
> 
> Are you using AI-Thinker ESP32-CAM with ESP32S written on the steel can?

closed 2022-10-07 by DJprasenjit

DJprasenjit, 2022-10-07:
no sir its just Esp32-S its has wifi logo bluetooth logo espressif logo and fccid only. not Ai thinker and logo in it.
but is not deffarent from the Ai thinker board. all looks same, and all works but the telegram function is not working, but another code the telegram bot working without any issue. 

can you help me , can you compile it and send me the bin file? i will upload it and i think it will work,
TIMEZONE "IST-5:30" 
const char* ssid = "FREE_WIFI-ZONE";
const char* password = "Password1";
#define BOTtoken  "5678841484:AAHTDDRXTYgm7n9wMdkRsq1tPZihBL75Ktw"   // get your own bot and id at telegram.org 
#define BOTme     "5108510368"
please sir its working with you. i am sure it will work. my telegram - @DJPrasenjit  or email- tantubaiprasenjit24@gmail.com