jameszah/ESP32-CAM-Video-Recorder #74: telegram sending error

DJprasenjit, 2022-10-08:
sir my post has been deleted. please help me fix this. i ordered a new esp32 cam ai thinker . i will try with that, the telegram is not working

jameszah, 2022-10-08:
https://github.com/jameszah/ESP32-CAM-Video-Recorder/issues/71#issue-1384765208

I'll do a configurable pre-compiled version.

DJprasenjit, 2022-10-09:
thank you sir. it will realy helpfull

DJprasenjit, 2022-10-09:
Sending photo by binary
z****Sending binary photo remaining buffer
*End request: 
--------------------------b8f610217e83e29b--
... Done Sending.  Client.Available = 0
zz... 2 secs later.  Client.Available = 1521

{"ok":true,"result":{"message_id":428,"from":{"id":5678841484,"is_bot":true,"first_name":"My ESP32Cam","username":"Prasenjit_bot"},"chat":{"id":5108510368,"first_name":"Redmi Note 4","username":"Krishna17704","type":"private"},"date":1665288021,"photo":[{"file_id":"AgACAgUAAxkDAAIBrGNCR1W_q-np2FD7gNmeMsiThJoDAAKbsTEbG60ZVgNGyqGpJofeAQADAgADcwADKgQ","file_unique_id":"AQADm7ExGxutGVZ4","file_size":604,"width":90,"height":67},{"file_id":"AgACAgUAAxkDAAIBrGNCR1W_q-np2FD7gNmeMsiThJoDAAKbsTEbG60ZVgNGyqGpJofeAQADAgADbQADKgQ","file_unique_id":"AQADm7ExGxutGVZy","file_size":5481,"width":320,"height":240},{"file_id":"AgACAgUAAxkDAAIBrGNCR1W_q-np2FD7gNmeMsiThJoDAAKbsTEbG60ZVgNGyqGpJofeAQADAgADeAADKgQ","file_unique_id":"AQADm7ExGxutGVZ9","file_size":8260,"width":640,"height":480}],"caption":"ESP32-CAM Video Recorder vA1\n desklens 192.168.0.104  \n Used 532 MB / 30424 MB, Rssi -43\n Sun Oct  9 09:30:18 2022\n\n Previous Video\n http://192.168.0.104:8080/c?dwn=/20221009/desklens_2022-10-09_09.29.50_vga_Q12_I100_L15_S1.avi","caption_entities":[{"offset":39,"length":13,"type":"url"},{"offset":133,"length":94,"type":"url"}]}}

Closing client

Send_photo heap after : 88684

Photo telegram was successfully sent 
z 
Total space: 30424MB
Used space: 532MB
sir i managed to send to telegram. now its working,, i changed the board to ai thinker esp32cam and now its working but filemanager gives error. the ftp was good. is the ftp still work on A1x version? and now board is over heating.

//20221009///20221009/desklens_2022-10-09_09.30.23_vga_Q12_I100_L15_S1.avi
Could not open file //20221009///20221009/desklens_2022-10-09_09.30.23_vga_Q12_I100_L15_S1.avi
//20221009///20221009/desklens_2022-10-09_09.30.18_vga_Q12_I100_L15_S1.jpg
Could not open file //20221009///20221009/desklens_2022-10-09_09.30.18_vga_Q12_I100_L15_S1.jpg
//20221009///20221009/desklens_2022-10-09_09.30.23_vga_Q12_I100_L15_S1.avi
Could not open file //20221009///20221009/desklens_2022-10-09_09.30.23_vga_Q12_I100_L15_S1.avi
//20221009///20221009/desklens_2022-10-09_09.30.23_vga_Q12_I100_L15_S1.avi
Could not open file //20221009///20221009/desklens_2022-10-09_09.30.23_vga_Q12_I100_L15_S1.avi

DJprasenjit, 2022-10-31:
while (Wait_for_bot == 1) {
        delay(1000);
        Serial.print("z");                      // serial monitor will shows these "z" mixed with "*" from telegram sender
      }

why is this delay for. its stops the code to run smoth, and telegram send fail, sometime success