jameszah/ESP32-CAM-Video-Recorder #77: wireless file manager download problem

DJprasenjit, 2022-12-30:
i am using latest version but in the file manager when i am going to download any file there is only the link comes with double Slashes and date twice
http://192.168.0.113:8080/c?dwn=**//20221226//**/20221226/desklens_2022-12-26_09.39.53_vga_Q12_I100_L15_S1.avi
to download i had to edit the link and remove this **//20221226//** every time. please tell me how to fix this problem.
thanks

lestitis, 2023-09-14:
i also have a problem deleting/renaming files
in the ESPxWebFIMgr.ccp i had to modify 
for rename
///        ESPxWebFlMgr_FileSystem.rename( subdir + "/" + fn,  subdir + "/" + fn2);
ESPxWebFlMgr_FileSystem.rename(fn, fn2);

for delete
///      int x = ESPxWebFlMgr_FileSystem.remove( subdir + "/" + fn);  // Add slash
      int x = ESPxWebFlMgr_FileSystem.remove(fn);  // Add slash

jameszah, 2023-09-15:
> i also have a problem deleting/renaming files in the ESPxWebFIMgr.ccp i had to 

I think that is a esp32-arduino version issue.  It had worked the original way, with the leading "/" implied, but then you had to provide it.  Maybe they have switched it back again.  What version do you have?

Tools -> Boards -> Board Manager -> "esp32"

Current is 2.0.13 --  I dont think Ive done much on this since 2.0,6

lestitis, 2023-09-16:
Hi, I have esp2.06Arduino ide 1.8.13 I bypassed the delete and rename problem, but not the download oneThe fn variable as the full path with twice the repository name.Thank you for your concern.Titi

lestitis, 2023-09-16:
Sorry, esp32 version 2.0.10Just installed 2.0.12