jameszah/ESP32-CAM-Video-Recorder #81: Compile error sys/_null.h file not exist

tonvanlee, 2024-01-01:
Hello,

I get error as below:
The problem is how to solve sys/_null.h. 
In order to compile I had already to download several libraries.
I also found the library _null.h but i have no idea where to store it.
I work with Arduino 2.2.1 and release V99.
I hope you can help me. Problem is also existing with older releases.

In file included from C:\Users\tonle\OneDrive\Documenten\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFiType.h:26,
                 from C:\Users\tonle\OneDrive\Documenten\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFi.h:33,
                 from C:\Users\tonle\OneDrive\Documenten\Arduino\libraries\Antares_ESP8266_HTTP\src\AntaresESP8266HTTP.h:5,
                 from C:\Users\tonle\OneDrive\Documenten\Arduino\libraries\Antares_ESP8266_HTTP\src\AntaresESP8266HTTP.cpp:1:
C:\Users\tonle\OneDrive\Documenten\Arduino\libraries\ESP8266WiFi\src/queue.h:38:10: fatal error: sys/_null.h: No such file or directory
 #include <sys/_null.h>
          ^~~~~~~~~~~~~
compilation terminated.
Multiple libraries were found for "WiFiClientSecure.h"
  Used: C:\Users\tonle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\libraries\WiFiClientSecure
  Not used: C:\Users\tonle\OneDrive\Documenten\Arduino\libraries\ESP8266WiFi
Multiple libraries were found for "WiFi.h"
  Used: C:\Users\tonle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\libraries\WiFi
  Not used: C:\Users\tonle\OneDrive\Documenten\Arduino\libraries\ESP8266WiFi
exit status 1

Compilation error: exit status 1

jameszah, 2024-01-01:
Those errors are for esp8266.
You need to set arduino board type to AI-Thinker ESP32-CAM

tonvanlee, 2024-01-01:
[image]
Also wit this board selected happens exactly the same error.
Maybe I have the wrong libraries installed??
Anyway thanks for the quick reply

jameszah, 2024-01-01:
You still have the 8266 libraries.  That null.h is an 8266 thing.
You might have to exit and restart the ide, or update the esp32 libraries.
[image]

tonvanlee, 2024-01-01:
I will remove temporary the esp8266wifi library. And then find the esp32wifi library.

image.png (view on web)

jameszah, 2024-01-01:
It is not the one wifi library ... the arduino compiler thinks you are writing esp8266 code, not esp32 code.
Once you get that sorted out, all the esp32 libraries come along for the ride.
You could try the Arduino 1.8.19 as more stable version.

tonvanlee, 2024-01-01:
Done compiling!!!I removed all esp8266wifi libraries from the library directory. Added wifimanager and arsuinojson and now compilation was correct. Thanks for the support. Fixed quite quickly. Bye,Ton

closed 2024-01-02 by tonvanlee